margelo/react-native-vision-camera #1215: Is it possible to do LIVE TEXT with react-native-vision-camera❓

iamabdulhaseeb, 2022-08-25:
### Question

I would like to implement a functionality that the user can place a camera on an image and he can copy text from image at that point, I have researched a lot but could not find any resource to do like this,
Is it possible to do this with this package?

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

justindavies, 2022-09-12:
I'm looking at this too - but it should be possible with a frame processor and the new Live Text things with IOS 16

iamabdulhaseeb, 2022-09-12:
Hey justin
If you are free these days
Can we both look together on this thing because I have been stucked on this
thing for days now.
I am creating my own app in which I wanted to implement this

marcelxsilva, 2022-11-26:
HI @iamabdulhaseeb,

I am using [`vision-camera-ocr`](https://github.com/aarongrider/vision-camera-ocr) to recognize text by camera, I don't understand how I can copy text but text can be shown in terminal easily

iamabdulhaseeb, 2022-11-26:
Hi @marcelxsilva , thank you for recommending that package, I will try that very soon :)

sampada-naik1998, 2023-04-20:
> HI @iamabdulhaseeb,
> 
> I am using [`vision-camera-ocr`](https://github.com/aarongrider/vision-camera-ocr) to recognize text by camera, I don't understand how I can copy text but text can be shown in terminal easily

can we use for ios?

marcelxsilva, 2023-04-20:
> > HI @iamabdulhaseeb,
> > I am using [`vision-camera-ocr`](https://github.com/aarongrider/vision-camera-ocr) to recognize text by camera, I don't understand how I can copy text but text can be shown in terminal easily
> 
> can we use for ios?

My app is only to Android, I believe that can use in iOS, without problems!!

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy