margelo/react-native-vision-camera #1216: Is this project deprecated❓

seriousStar, 2022-08-25:
### Question

Is this project deprecated❓ 

### What I tried

I tried to post some questions, so far no updates yet.

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

kicksent, 2022-09-03:
No, the creator is still posting comments on PRs so it doesn't seem to be deprecated.

mrousavy, 2023-09-30:
Lmao, nope.

closed 2023-09-30 by mrousavy