margelo/react-native-vision-camera #1221: 🐛 vision-camera-face-detector Is not updated for a long time

DeepikaSharma5, 2022-08-31:
### What were you trying to do?

I am trying to implement an app with camera and face detection. As I have checked your documentation you have mentioned that we can use vision-camera-face-detector the detect the face.

I used that and able to detect the face in android devices. But the face detection is not working for IOS. 

The package is not well maintained and the npm packages is not updated. Even there is no one to answer or resolve the bugs. 

Please takeover the package and try to fix the issues. This will be a big help for a lot of us. 

### What happened instead?

Face detection is not working in IOS devices

### Device

iPhone 12 Pro max

### VisionCamera Version

Latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

AdityaVJ, 2022-09-02:
I've faced the same issue in one of my projects in which I needed to use the face-detector module. The library integration unfortunately, is no longer maintained by the developer. The last release at the time of writing this was on 15th of October 2021 (version 0.1.8).

You will have to pull in their library by pointing to their master branch by doing something like:
`"vision-camera-face-detector": "https://github.com/rodgomesc/vision-camera-face-detector#master"`

in your package.json. I know it's a hack. The other way would be to fork the master in the current state and create a release on your own.

DeepikaSharma5, 2022-09-05:
Yes I am planning to fork and continue the project. But it's better if react-native-vision-camera team maintains that package. Other wise its better to remove the package form the document (community plugin list)

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy