margelo/react-native-vision-camera #1223: Integrating tfjs-react-native❓

ShivaAryal, 2022-09-02:
### Question

Is it possible to integrate tfjs-react-native(https://github.com/tensorflow/tfjs/tree/master/tfjs-react-native) when react-native-vision-camera is used in the app as tfjs-react-native uses expo-camera by default❓ 

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

eledahl, 2023-01-14:
+10000000000

fukemy, 2023-06-19:
read https://github.com/mrousavy/react-native-vision-camera/issues/1073

mrousavy, 2023-06-22:
See https://github.com/tensorflow/tfjs/issues/7773 :)

mrousavy, 2023-09-30:
I built a JSI/C++/GPU-accelerated library just for this: https://github.com/mrousavy/react-native-fast-tflite 🥳

closed 2023-09-30 by mrousavy