margelo/react-native-vision-camera #1224: 🐛 Barcode is not recognized correctly

jacob-li-hj, 2022-09-02:
### What were you trying to do?

The scanning speed of the camera is relatively fast, but there is a certain chance that the barcode will be recognized as wrong data. I tried to use the provided demo to test, but there is still such a problem.

### Reproduceable Code

```tsx
"react": "17.0.2",
"react-native": "0.66.3",
"react-native-reanimated": "2.4.1",
"react-native-screens": "3.5.0",
"react-native-vision-camera": "^2.9.4"

import * as React from 'react';

import { StyleSheet } from 'react-native';
import { useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
import { Camera } from 'react-native-vision-camera';
import { useScanBarcodes, BarcodeFormat, scanBarcodes } from 'vision-camera-code-scanner';

export default function App() {
  const [hasPermission, setHasPermission] = React.useState(false);
  const devices = useCameraDevices();
  const device = devices.back;

  const frameProcessor = useFrameProcessor(
    frame => {
      // eslint-disable-next-line
      'worklet'
      const barcodes = scanBarcodes(frame, [BarcodeFormat.ALL_FORMATS])
      if(barcodes[0]?.content?.data){
        console.log('111111111111', barcodes[0]?.content?.data)
      }
    },
    []
  )

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  return (
    device != null &&
    hasPermission && (
      <>
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          frameProcessor={frameProcessor}
          frameProcessorFps={5}
        />
      </>
    )
  );
}
```

### What happened instead?

Barcodes tested:Code-39 type barcode data generated by https://barcode.tec-it.com/en/Code39FullASCII?data=32343242 website：
[image]
But there is a chance that it will be recognized as wrong data

[image]

### Device

All devices have a chance to reproduce

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2022-09-05:
Is your camera stable? The barcode image may be blurred which may cause misreading.

You can also turn to this plugin using Dynamsoft Barcode Reader: <https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader>, which is more robust.

jacob-li-hj, 2022-09-07:
> Is your camera stable? The barcode image may be blurred which may cause misreading.
> 
> You can also turn to this plugin using Dynamsoft Barcode Reader: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader, which is more robust.

https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader I tested this, it's good. There is no case of scanning the code and showing wrong data now, but this SDK is too expensive. Do you have any other recommendations, thanks.

Productivix, 2022-09-12:
hi , the problem comes from your barcode printed : I cannot  read the 39-barcode you give here with a professionnal barcode terminal : so work on the barcode print first (the web site you named provide codes in .gif without the quiet zone first) - but if use the 128 font instead of 39 is ok if you make a screen shot copy from their site.

jacob-li-hj, 2022-09-12:
> 

I'm just giving an example, the barcodes that can have issue are not just code-39 type barcodes. But barcodes in real products also have this issue. According to what you said, we have tested the code-128 type barcode and did not find this issue for the time being, but there are not only code-128 type barcodes in the real situation.

Productivix, 2022-09-13:
normal : I got some distortions with 39 also : there is no checksum in 39 font, there is on I25, 128 fonts. Do not use 39 in production.

jacob-li-hj, 2022-09-14:
> normal : I got some distortions with 39 also : there is no checksum in 39 font, there is on I25, 128 fonts. Do not use 39 in production.

Thank you for your answer, but we need to support most types of barcode, if you have any solution please let me know, thank you very much.

jacob-li-hj, 2022-09-15:
> normal : I got some distortions with 39 also : there is no checksum in 39 font, there is on I25, 128 fonts. Do not use 39 in production.

https://github.com/react-native-camera/react-native-camera/tree/master, I did not find this problem when using react-native-camera, but react-native-camera scan code is too slow. After encountering this problem, I have to continue to use react-native-camera. The difference between the two is that one uses Firebase/MLVision and the other uses GoogleMLKit/BarcodeScanning. I haven't studied it in detail yet, and I don't know why this problem occurs.

Productivix, 2022-09-15:
I think it's better that you close that issue here and continue on https://github.com/rodgomesc/vision-camera-code-scanner/issues/95

jacob-li-hj, 2022-09-15:
>

closed 2022-09-15 by jacob-li-hj