margelo/react-native-vision-camera #1228: Photos took after motion are blurred + lack of autofocus

adkrk, 2022-09-06:
### Question

I get blurred photos when I take a photo immediately after moving the phone. This does not happen with my default Samsung camera app.

### What I tried

I have tried to use every format that matches the resolution I have set in my phone's camera app. Surprisingly, every format has: "autoFocusSystem": "none", even though the default camera app has autofocus.

When I take pictures completely statically through the library, they are not blurred, but I'm not getting the same kind of focus effect. I'm guessing it could be that the camera does not have autofocus at hardware level and the camera app implements it programmatically, so when I try to use the camera library in a custom app, I have no autofocus and the blur after motion could also be caused by a lack of some programmatic secret sauce that the default camera app implements?

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

GreenCurveDev, 2022-09-14:
Hi, 
I have also noticed the same issue on an iPhone 12 - V 2.14.1. (ie blurred image with slight phone movement - not seen on native camera). Am not using Expo.
Couldn't find the equivalent of VideoStabilizationMode for still images but have set this anyway to 'auto' (no impact on stills)
Also, tap to focus also does not seem to work (followed guide in documentation) but would be happy if auto focus did.
Appreciate any help with resolving this issue - it's the only one encountered so far and otherwise this package is great!

awcchungster, 2022-09-15:
You have to autofocus on a point. For my use case, I always take the center of the photo. Here's a code snippet:

```
const centerPoint = {
          x: Dimensions.get('window').width / 2,
          y: Dimensions.get('window').height / 2,
};
console.debug(`Autofocusing on center`);
return cameraRef.current.focus(centerPoint);
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

dilawer-dev, 2023-10-05:
I am facing the same issue Image taken by native camera is less blur if my hand is shaky and if I capture the image using this package then pictures or blur if my hand is shaky