margelo/react-native-vision-camera #1239: 🐛 work on debug version but not work on release

MarquisAlexander, 2022-09-12:
### What were you trying to do?

Open Camera on my devices (xiaomi redmi 6, Samsung s21 and emulator Pixel 3a API 29)

### Reproduceable Code

```tsx
import React, { useState } from 'react';
import { Text, TouchableOpacity, View } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

function App() {
  const devices = useCameraDevices();
  const device = devices.back;

  const [hasPermission, setHasPermission] = useState(false);

  async function getPermissions() {
    const newCameraPermission = await Camera.requestCameraPermission();
    const newMicrophonePermission = await Camera.requestMicrophonePermission();
    setHasPermission(
      newCameraPermission === 'authorized' && newMicrophonePermission === 'authorized'
    );
  }

  return (
    <View
      style={{
        flex: 1,
        alignItems: 'center',
        justifyContent: 'center',
      }}
    >
      <TouchableOpacity onPress={() => getPermissions()}>
        <Text>turn on camera</Text>
      </TouchableOpacity>
      {hasPermission && device != null ? <Text>camera is on</Text> : <Text>down camera</Text>}
    </View>
  );
}
export default App;
```

### What happened instead?

When I'm in the debug environment with **yarn react-native run-android --variant=debug**, **useCameraDevices** returns all the necessary information ex:
```
{
   "back":{
      "devices": ["telephoto-camera"],
      "formats":[
         [
            "Object"
         ]...
      ],
      "hasFlash":false,
      "hasTorch":false,
      "id":"0",
      "isMultiCam":false,
      "maxZoom":10,
      "minZoom":1,
      "name":"back (0)",
      "neutralZoom":1,
      "position":"back",
      "supportsDepthCapture":false,
      "supportsFocus":true,
      "supportsLowLightBoost":false,
      "supportsParallelVideoProcessing":false,
      "supportsRawCapture":false
   },
   "external":"undefined",
   "front":{
      "devices":["telephoto-camera"],
      "formats":[
         [
            "Object"
         ]...
      ],
      "hasFlash":false,
      "hasTorch":false,
      "id":"1",
      "isMultiCam":false,
      "maxZoom":10,
      "minZoom":1,
      "name":"front (1)",
      "neutralZoom":1,
      "position":"front",
      "supportsDepthCapture":false,
      "supportsFocus":true,
      "supportsLowLightBoost":false,
      "supportsParallelVideoProcessing":true,
      "supportsRawCapture":false
   },
   "unspecified":"undefined"
}
```

but when I'm in the release environment with **yarn react-native run-android --variant=release**, **useCameraDevices** is undefined, ex:
```
{
   "back":{"undefined",
   "external":"undefined",
   "front":"undefined",
   "unspecified":"undefined"
}
```

**_Relevant info_** 
**android/build.gradle**
```
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 24
        compileSdkVersion = 31
        targetSdkVersion = 31
        kotlinVersion = '1.5.0'
        ndkVersion = "20.1.5948944"
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.2.1")
    }
}
```

**android/gradle/wrapper/gradle-wrapper.properties**
`distributionUrl=https\://services.gradle.org/distributions/gradle-6.9-all.zip`

### Device

xiaomi redmi 6, Samsung s21 and emulator Pixel 3a API 29

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2022-09-13 by MarquisAlexander

labiebhn, 2022-10-29:
how do you solve this issue?

ashrazdev, 2022-11-30:
how do you solve this issue?