margelo/react-native-vision-camera #1241: detect shapes (circular, square)✨

Productivix, 2022-09-13:
### What feature or enhancement are you suggesting?

I think that process can detect geometric shapes as disks, circles, squares ?
Would be useful to count rapidly materials as tubes , bars from a profile photo

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

not known apart from Google..

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

Productivix, 2023-09-30:
??

mrousavy, 2023-09-30:
❓❓