margelo/react-native-vision-camera #1243: ❓ when takePhoto, report error capture/file-io-error

zhuernm, 2022-09-14:
### Question

First time: camera.current.takePhoto succeed;
Second time: reload Camera, when takePhoto(), report error " capture/file-io-error "

### What I tried

tried RN 0.67 and 0.68. Both failed 
tried  cleaning and rebuilding everything, failed

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zhuernm, 2022-09-14:
Wierd

Got the cause, the folder is locked. But I don't know why.

//TODO 清理缓存目录
    // try {
    //   const pro = [];
    //   const allf = await RNFS.readDir(RNFS.CachesDirectoryPath);
    //   console.log('所有文件', allf && allf.length);
    //   if (allf && allf.length) allf.forEach(ff => pro.push(RNFS.unlink(ff.path)));

    //   const allf2 = await RNFS.readDir(RNFS.TemporaryDirectoryPath);
    //   console.log('所有文件2', allf2 && allf2.length);
    //   if (allf2 && allf2.length) allf2.forEach(ff => pro.push(RNFS.unlink(ff.path)));
    //   await Promise.all(pro);
    // } catch (err) { console.log('清理。失败==', err) }

closed 2022-09-14 by zhuernm

adscud, 2023-07-07:
@moonofmars How did you fix this issue ?