margelo/react-native-vision-camera #1244: 🐛 Vision Camera crashes when built with Xcode 14

jpudysz, 2022-09-15:
### What were you trying to do?

I was trying to build my app with Xcode 14.

### Reproduceable Code

```tsx
No super fancy code. Simply render the camera with a frame processor. I'm using QR code scanner:

const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.CODE_128])

return (
    <Camera
       device={device}
       isActive={isActive}
       frameProcessorFps={5}
       frameProcessor={frameProcessor}
       style={StyleSheet.absoluteFillObject}
    />
)
```

### What happened instead?

The app crashed with an error from HostFunction.

I can confirm that code works as expected when built with Xcode 13.

[image]

### Relevant log output

```shell
2022-09-15 10:13:26.797643+0200 x[6471:486752] [native] VisionCamera.didSetProps(_:): Updating 13 prop(s)...
2022-09-15 10:13:26.797882+0200 x[6471:487061] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2022-09-15 10:13:26.797894+0200 x[6471:487061] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:0...
2022-09-15 10:13:26.797906+0200 x[6471:487061] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2022-09-15 10:13:26.798369+0200 x[6471:487075] FrameProcessorBindings: Setting new frame processor...
2022-09-15 10:13:26.798393+0200 x[6471:487075] FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...
2022-09-15 10:13:26.798517+0200 x[6471:487075] FrameProcessorBindings: Successfully created worklet!
2022-09-15 10:13:26.798554+0200 x[6471:487097] FrameProcessorBindings: Converting worklet to Objective-C callback...
2022-09-15 10:13:26.799270+0200 x[6471:487097] FrameProcessorBindings: Frame processor set!
2022-09-15 10:13:26.838111+0200 x[6471:487061] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2022-09-15 10:13:26.838707+0200 x[6471:487061] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2022-09-15 10:13:26.838926+0200 x[6471:487061] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2022-09-15 10:13:26.839396+0200 x[6471:487061] [native] VisionCamera.configureFormat(): Configuring Format...
2022-09-15 10:13:26.839401+0200 x[6471:487061] [native] VisionCamera.configureDevice(): Configuring Device...
2022-09-15 10:13:26.839427+0200 x[6471:487061] [native] VisionCamera.configureDevice(): Device successfully configured!
2022-09-15 10:13:26.839689+0200 x[6471:487061] [native] VisionCamera.didSetProps(_:): Starting Session...
2022-09-15 10:13:27.007614+0200 x[6471:487061] [native] VisionCamera.didSetProps(_:): Started Session!
libc++abi: terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
(Recorded stack frame) 

I've debugged the code and it crashes in FrameProcessorUtils.mm:

cb.callWithThis(runtime, cb, jsi::Object::createFromHostObject(runtime, frameHostObject));
```

Null pointer exception
```

### Device

iPhone 13

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

baylesa-dev, 2022-09-15:
Hello @jpudysz, could you add your react-native/react-native-vision-camera versions please ? :)

jpudysz, 2022-09-15:
Sure!
```
"react-native": "0.69.5",
"react-native-vision-camera": "2.14.1",
"vision-camera-code-scanner": "0.2.0"
```

icc-guerrero, 2022-09-15:
Similar issue here. 

[javascript] Frame Processor threw an error: Can't find variable: __xxxx

baylesa-dev, 2022-09-15:
I'm able to reproduce the issue (repro https://github.com/baylesa-dev/rn_vision_camera_xcode_14)

jpudysz, 2022-09-15:
Nice catch @baylesa-dev!
The problem is that we can't even call this plugin. It breaks before `callback`:
https://github.com/rodgomesc/vision-camera-code-scanner/blob/0571128d46a4a7564aca97d9271c8f7c0b67a5d7/ios/VisionCameraCodeScanner.swift#L11

baylesa-dev, 2022-09-15:
Yep! 
```ts
console.log(typeof __labelImage) // => function
console.log(typeof __scanCodes) // => undefined
```

jpudysz, 2022-09-15:
I also checked [vision-camera-face-detector](https://github.com/rodgomesc/vision-camera-face-detector), and it crashed immediately. The only difference between all these plugins is that `vision-camera-image-labeler` has been written with Objective-C instead of Swift. 

I suspect that `VISION_EXPORT_SWIFT_FRAME_PROCESSOR` is not registering frame processors correctly or something else breaks this registration in Xcode 14.

sacmii, 2022-09-15:
npx vision-camera-plugin-builder ios => Objective C Language
[image]

Same issue

baylesa-dev, 2022-09-15:
Mentioned in https://github.com/facebook/react-native/issues/34673#issuecomment-1245377640

jpudysz, 2022-09-16:
So far, nothing I'm aware of.
I was trying to patch Podfile with the solutions mentioned in the linked issue, and the app is compiling indeed, but vision camera plugins are not working.

We need to wait for a patch from react-native.
To run your app smoothly, you need to downgrade Xcode to version 13.4.

bglgwyng, 2022-09-18:
I have this issue after upgrading RN to 0.70.1

Neklan, 2022-09-22:
> So far, nothing I'm aware of. I was trying to patch Podfile with the solutions mentioned in the linked issue, and the app is compiling indeed, but vision camera plugins are not working.
>
> We need to [...]

Thanks, I confirm that this solution work, for me is just temporary workaround, because I need to deploy to the latest iOS, so I had to downgrade to Xcode 13.4 and then add support for iOS 16.0

Hope this will be fixed also for the latest Xcode version

In this thread you can find, how to add support

https://developer.apple.com/forums/thread/673131

[image]

intmain, 2022-09-28:
VISION_EXPORT_SWIFT_FRAME_PROCESSOR is working on my main Project, but not working in pod project like vision-camera-code-scanner when build by Xcode 14.0.

If I chagne this code `__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()` to `+(void)load`

then it work (temporary). I tested it.

EasonLex, 2022-09-29:
> VISION_EXPORT_SWIFT_FRAME_PROCESSOR is working on my main Project, but not working in pod project like vision-camera-code-scanner when build by Xcode 14.0.
>
> If I chagne this code [...]

Thanks @intmain , it fixed the issue for me temporarily too. It seems that using __attribute__((constructor)) no longer works with load() and not sure why swift bridge to objc works (using +(void)load )

(I'm not familiar with swift, so if I was wrong with something, please correct me.)

Neklan, 2022-09-30:
> VISION_EXPORT_SWIFT_FRAME_PROCESSOR

Can you please provide more information on how to modify the code and where, so we can apply it also? Thank you

bglgwyng, 2022-10-01:
@Neklan 
You can apply that patch with the following patch script and [patch-package](https://www.npmjs.com/package/patch-package),

```diff
diff --git a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h
index a2ccdcb4..7c9690e0 100644
--- a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
+++ b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
@@ -53,7 +53,7 @@ objc_name : NSObject<FrameProcessorPluginBase>
 @end                                                                                \
 @implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                     \
-__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
++(void)load                                                                         \
 {                                                                                   \
   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
     return [objc_name callback:frame withArgs:args];                               \
```

Sadly, it didn't fix my problem with custom frame processor https://github.com/mrousavy/react-native-vision-camera/issues/1244#issuecomment-1250233793. Or those problems are irrelevant to each other.

zzz08900, 2022-10-01:
> VISION_EXPORT_SWIFT_FRAME_PROCESSOR is working on my main Project, but not working in pod project like vision-camera-code-scanner when build by Xcode 14.0.
>
> If I chagne this code [...]

Thanks man, issue fixed too.

zzz08900, 2022-10-01:
> @Neklan You can apply that patch with the following patch script and [patch-package](https://www.npmjs.com/package/patch-package),
> 
> ```diff
> diff --git a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h
> index a2ccdcb4..7c9690e0 100644
> --- a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
> +++ b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
> @@ -53,7 +53,7 @@ objc_name : NSObject<FrameProcessorPluginBase>
>  @end                                                                                \
>  @implementation objc_name (FrameProcessorPlugin)                                    \
>                                                                                      \
> -__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
> ++(void)load                                                                         \
>  {                                                                                   \
>    [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
>      return [objc_name callback:frame withArgs:args];                               \
> ```
> 
> Sadly, it didn't fix my problem with custom frame processor [#1244 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/1244#issuecomment-1250233793). Or those problems are irrelevant to each other.

The method posted by @intmain only solved crash caused by Xcode 14 - which means if you're having problems with Xcode 13.x, then the problem is probably elsewhere.

Neklan, 2022-10-03:
I confirm, that solution posted above with changing `__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()` to `+(void)load` is working on the newest xCode 14.0.1 with custom frame processor

alexco2, 2022-10-03:
> @Neklan You can apply that patch with the following patch script and [patch-package](https://www.npmjs.com/package/patch-package),
>
> ```diff
> diff --git [...]

this fixed the issue for me as well

SlowDeath7, 2022-10-10:
Same issue in android, please help me .  Have switched back to react native camera which is deprecated.

ikuruzum, 2022-10-10:
> @Neklan You can apply that patch with the following patch script and [patch-package](https://www.npmjs.com/package/patch-package),
>
> ```diff
> diff --git [...]

how can i make and use that script

zzz08900, 2022-10-11:
@ikuruzum all you need to do is googling "how to use patch-package"

zzz08900, 2022-10-11:
> Same issue in android, please help me . Have switched back to react native camera which is deprecated.

This issue has nothing to do with android, you're facing a different problem

qtipee, 2022-10-18:
I confirm that using the patch above works with Xcode 14.0.1 :)

I had no problem when using Objective-C for the iOS native module, but as soon as you use Swift, you need to apply that patch.

LazyAfternoons, 2022-10-19:
@mrousavy can you please look at this? Seems to work fine.

mrousavy, 2022-10-20:
Huh, that's interesting. I'll take a look to find out if that's safe to do

euthribeiro, 2022-10-24:
@intmain thank you so much!

LazyAfternoons, 2022-10-27:
> Could you create a PR with the change?

Hey @mrousavy I created the PR with the change proposed by @intmain #1308.
Thank you!

closed 2022-10-28 by mrousavy

mehtaanirudh, 2022-12-22:
Do not use this library if you are using `React Native > 0.68` and `Firebase > 15.0`

There are many issues and I am not sure if this library is actively maintained. You may want to use `react-native-ml-kit`.

I have not tried to use `react-native-ml-kit` with frame processors but you can use `rn-vision-camera` to take the picture and pass the image to `react-native-ml-kit` to get the bar code. Example app is here -> https://github.com/a7medev/react-native-ml-kit/tree/main/barcode-scanning/example

I wasted a day trying to get this library to work with 0.69.3 but could not so if anyone is interested then they can use `react-native-ml-kit`

TacticCoder, 2022-12-27:
Another way around it, remove use_framework! and provide the pods manually.

marcshilling, 2023-09-13:
So, this change makes my app crash at launch when built using Xcode 15...The method `Swift class extensions and categories on Swift classes are not allowed to have +load methods` gets logged, right before this crash:

```
Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: OBJC 1 

Triggered by Thread:  0

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x11356da52 __abort_with_payload + 10
1   libsystem_kernel.dylib        	       0x11358b84e abort_with_payload_wrapper_internal + 82
2   libsystem_kernel.dylib        	       0x11358b7fc abort_with_reason + 19
3   libobjc.A.dylib               	    0x7ff800057593 _objc_fatalv(unsigned long long, unsigned long long, char const*, __va_list_tag*) + 121
4   libobjc.A.dylib               	    0x7ff80005751a _objc_fatal(char const*, ...) + 114
5   libobjc.A.dylib               	    0x7ff80004166d load_images + 1383
6   dyld_sim                      	       0x1120b909b dyld4::RuntimeState::notifyObjCInit(dyld4::Loader const*) + 359
7   dyld_sim                      	       0x1120be73c dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 182
8   dyld_sim                      	       0x1120c149a dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const::$_1::operator()() const + 92
9   dyld_sim                      	       0x1120be7dd dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 93
10  dyld_sim                      	       0x1120d43df dyld4::APIs::runAllInitializersForMain() + 263
11  dyld_sim                      	       0x1120b010a dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 2535
12  dyld_sim                      	       0x1120af356 _dyld_sim_prepare + 854
13  dyld                          	       0x119570424 dyld4::prepareSim(dyld4::RuntimeState&, char const*) + 1490
14  dyld                          	       0x11956eabc dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 249
15  dyld                          	       0x11956e3bd start + 1805
```

PR #1308 (2022-10-28): fix: crash when using a frame processor on Xcode 14

## What
This PR fixes a crash when using a frame processor on Xcode 14.

## Changes
This solution was proposed by @intmain in the issue discussion.

## Tested on
iPhone XS, iOS 16.0.2.

## Related issues
Fixes #1244