margelo/react-native-vision-camera #1245: 🐛 Autofocus returning errors on newest Android devices

awcchungster, 2022-09-15:
### What were you trying to do?

I want to autofocus on the center of the video on an Android device. IF the device does not support autofocus, I want it to gracefully terminate. I add an if statement that checks the support for autofocus before actually calling the autofocus function.

https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDevice#supportsfocus

### Reproduceable Code

```tsx
With a camera reference instantiated

        if (camera?.supportsFocus) {
        console.log('supports focus');
        // grab the centerpoint to autofocus on
        const centerPoint = {
          x: Dimensions.get('window').width / 2,
          y: Dimensions.get('window').height / 2,
        };

        console.debug(`Autofocusing on center`);
        return cameraRef.current.focus(centerPoint);
        }
```
```

### What happened instead?

The device waits for the focus function for several seconds and then spits the following error. This behavior is Android specific

### Relevant log output

```shell
This device does not support autofocus:  [unknown/unknown]: [unknown/unknown] Cancelled by cancelFocusAndMetering()
```

### Device

Samsung s22

### VisionCamera Version

1.0.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

wilav-dev, 2022-09-30:
Similar issue: Cancelled by another startFocusAndMetering()

scaralfred, 2022-10-25:
Are there any plan to fix this?

andreiwow2, 2023-01-09:
If I try this on iphone 12 the app freezes and I can't even reload it from bundle

osbornZ, 2023-07-18:
Are this issue fixed？
I got this similar ：
[unknown/unknown] None of the specified AF/AE/AWB MeteringPoints is supported on this camera.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy