margelo/react-native-vision-camera #1248: 🐛 react-native-vision-camera Crash on MI

msen04377, 2022-09-17:
### What were you trying to do?

                      <>
                        <Camera
                          ref={camera}
                          device={device}
                          style={StyleSheet.absoluteFill}
                          isActive={true}
                          video={true}
                          audio={true} />
                      </>

### Reproduceable Code

```tsx
<>
                        <Camera
                          ref={camera}
                          device={device}
                          style={StyleSheet.absoluteFill}
                          isActive={true}
                          video={true}
                          audio={true} />
                      </>

if i am doing isActive = true then it is crashed when i am recording the video in MI and REALME phone althought other phone like samsung , oppo it is working fine.
```

### What happened instead?

No error getting , even i am applying adb logcat but no any i am getting

### Device

RMX3085 Reakme 8

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hasnainbaloch, 2022-09-17:
Facing same issue any solution?

hasnainbaloch, 2022-09-17:
java.lang.IllegalStateException: Surface was requested when the Recorder had encountered error androidx.camera.video.internal.ResourceCreationException: androidx.camera.video.internal.encoder.InvalidConfigException
.
.
.

hasnainbaloch, 2022-09-17:
Camera was working fine before the new **Android** update and this package is still working fine in **IOS**

zzz08900, 2022-10-01:
> Camera was working fine before the new **Android** update and this package is still working fine in **IOS**

Mind to elaborate what  "new Android update" are we talking about here? Update from Android 11 to 12 or just some random OTA patch issued by phone manufacturers?

xeroxoid, 2023-01-24:
Possible duplicate of #907

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy