margelo/react-native-vision-camera #1249: Why boost and some other files are always downloaded for react native vision camera whenever I run the app after changes to package.json❓

MuhammadAbdullah54321, 2022-09-19:
### Question

So whenever I install or delete any new package and run `yarn android`, boost is started downloading.
Boost is downloaded from this link: https://github.com/react-native-community/boost-for-react-native/releases/download/v1.63.0-0/boost_1_63_0.tar.gz

Some other things are also downloaded every-time like doubleConversion
Link used is: https://github.com/google/double-conversion/archive/v1.1.6.tar.gz

I know it's not huge files but still It's kind of annoying. 

I am using 
React Native version: `0.63.4`
React version: `16.13.1`
Gradle version: `6.7.1`
Android Build tools with gradle version: `4.2.2`

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

kwangjaeyoo, 2022-10-17:
i have same issue ....

my info

System:
    OS: macOS 12.5.1
    CPU: (10) arm64 Apple M1 Pro
    Memory: 765.83 MB / 32.00 GB
    Shell: 5.8.1 - /bin/zsh
  Binaries:
    Node: 18.10.0 - /usr/local/bin/node
    Yarn: 1.22.19 - /opt/homebrew/bin/yarn
    npm: 8.19.2 - /opt/homebrew/bin/npm
    Watchman: 2022.10.03.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.11.3 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 21.4, iOS 15.5, macOS 12.3, tvOS 15.4, watchOS 8.5
    Android SDK: Not Found
  IDEs:
    Android Studio: 2021.2 AI-212.5712.43.2112.8815526
    Xcode: 13.4.1/13F100 - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.16.1 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: ^9.1.3 => 9.1.3 
    react: 18.1.0 => 18.1.0 
    react-native: 0.70.3 => 0.70.3 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy