margelo/react-native-vision-camera #1251: 🐛 Android Black Screen

Easy47, 2022-09-19:
### What were you trying to do?

When I launch the example app, it just gives me a black screen. I can still press the buttons for flash and for swap but the flash does not work either and the swap does not remove the black screen.(I don't have this problem with version 2.13, only with 2.14).

It seems that I have the permissions and I have no error message, just a black screen on my phone with the buttons. I tried this https://github.com/mrousavy/react-native-vision-camera/issues/879 too but it the same.

I'm starting to wonder if this is a bug with 2.14 ? I have gone through the issues and everyone seems to have errors with the black screen.

Any idea ?
Thanks

### Reproduceable Code

```tsx
import * as React from 'react';
import { useRef, useState, useMemo, useCallback, useEffect } from 'react';

import { useIsFocused } from '@react-navigation/native';
import { StyleSheet, View, PermissionsAndroid } from 'react-native';
import { PinchGestureHandler, PinchGestureHandlerGestureEvent, TapGestureHandler, TouchableOpacity } from 'react-native-gesture-handler';
import Reanimated, { Extrapolate, interpolate, useAnimatedGestureHandler, useAnimatedProps, useSharedValue } from 'react-native-reanimated';
import IonIcon from 'react-native-vector-icons/Ionicons';
import {
  CameraDeviceFormat,
  PhotoFile,
  sortFormats,
  useCameraDevices,
  Camera as Cam,
  frameRateIncluded,
  VideoFile,
  useFrameProcessor
} from 'react-native-vision-camera';
import type { NativeStackScreenProps } from '@react-navigation/native-stack';
import type { Routes } from './Routes';
import { CONTENT_SPACING, SAFE_AREA_PADDING } from './Constants';
import { CaptureButton } from './views/CaptureButton';

import { scanOCR } from 'vision-camera-ocr';
const ReanimatedCamera = Reanimated.createAnimatedComponent(Cam);
Reanimated.addWhitelistedNativeProps({
  zoom: true,
});

const BUTTON_SIZE = 40;
const SCALE_FULL_ZOOM = 3;

type Props = NativeStackScreenProps<Routes, 'CameraPage'>;
export function CameraPage({ navigation }: Props): React.ReactElement {
  const camera = useRef<Cam>(null);
  const [isCameraInitialized, setIsCameraInitialized] = useState(false);
  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back');
  const [enableNightMode, setEnableNightMode] = useState(false);
  const [flash, setFlash] = useState<'off' | 'on'>('off');
  const isPressingButton = useSharedValue(false);

  const devices = useCameraDevices();
  const zoom = useSharedValue(0);
  const isFocussed = useIsFocused();

  const isActive = isFocussed;
  const device = devices[cameraPosition];

  const minZoom = device?.minZoom ?? 1;
  const maxZoom = Math.min(device?.maxZoom ?? 1, 20);
  const neutralZoom = device?.neutralZoom ?? 1;
  const supportsFlash = device?.hasFlash ?? false;

  const onMediaCaptured = useCallback(
    (media: PhotoFile | VideoFile, type: 'photo' | 'video') => {
      console.log(`Media captured! ${JSON.stringify(media)}`);
[... 174 lines cut ...]

### What happened instead?

Black screen

### Relevant log output

```shell
No output :/ Seems like everything works but I still have a blackscreen !
```

### Device

Android

### VisionCamera Version

2.14.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Easy47, 2022-09-19:
I figured that if i delete this line "format={format}" i don't have the black screen anymore ... don't know why

mrousavy, 2022-10-19:
Hey, thanks for your issue report!

## Problem

Back in 2021 I developed VisionCamera around iOS' native device / format API. This Swift API is really flexible and allows you to get a list of all devices, **and each device has a list of all supported formats**. That's exactly how VisionCamera exposes it to JavaScript as well.

On Android however, I use CameraX. CameraX is the new recommended native camera library for Android. It does a lot of abstractions already, and while it's possible to get a list of all camera devices (front, back, wide angle, ..), it doesn't tell you which format is supported on which device. There's no such thing as "formats" in CameraX, you kinda just tell CameraX what resolution you want to record in, and it tries to match that as good as it can. All of the filtering is done internally then.

This sucks, because that's the same abstraction that I built using VisionCamera, and now it's not really possible to expose a list of formats through JavaScript, since I kinda have to make them up and just guess that it works, this is done here: https://github.com/mrousavy/react-native-vision-camera/blob/4781ad983530493ee13069fd69ef9f15d154a3cc/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L168-L342

## Solution

I've been thinking about this for a long time now. I want to fix this issue once and for all with **VisionCamera v3**.

V3 has these features planned:

1. Will be a Fabric/TurboModule only. Legacy architecture will not be supported for the sake of simplicity (will fix a ton of build errors)
2. Write-back frame processors (draw on frame using RN Skia, draw face masks, filters, overlays, color shadings, shaders, Metal, etc..)
3. Completely redesigned API for device/format selection (resolution, fps, low-light, ...)

I will brainstorm with @thomas-coldwell in our Vienna Margelo office to think about how we can achieve a perfect API redesign that works on both iOS and Android without compromising too much of the flexibility that we have today.
This means, the new API should still allow things like:

- Control exactly how much FPS you want to record in
- Know exactly if a desired format is supported and be able to fall back to a different one
- Control the exact resolution and know what is supported (e.g. higher than 1080, but no higher than 4k, ...)
- Control settings like low-light mode, compression, recording format H.264 or H.265, etc. 

For this we need a lot of time though.

If you (or your company) is interested in financially supporting those efforts, please let me know. Without sponsorship we can't work on this.

mrousavy, 2022-12-22:
Btw., V3 discussion is here: https://github.com/mrousavy/react-native-vision-camera/issues/1376

closed 2023-09-01 by mrousavy