margelo/react-native-vision-camera #1253: 🐛

lichmac, 2022-09-23:
### What were you trying to do?

I want to add the QRcode scanner. iOS works fine, but Android compile failed

### Reproduceable Code

```tsx
cd android & ./gradlew clean & cd .. & npx react-native run-android

System:
    OS: macOS 12.4
    CPU: (8) arm64 Apple M2
    Memory: 94.59 MB / 16.00 GB
    Shell: 5.8.1 - /bin/zsh
  Binaries:
    Node: 18.7.0 - ~/.nvm/versions/node/v18.7.0/bin/node
    Yarn: 1.22.19 - ~/.nvm/versions/node/v18.7.0/bin/yarn
    npm: 8.15.0 - ~/.nvm/versions/node/v18.7.0/bin/npm
    Watchman: 2022.08.15.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.11.3 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 21.4, iOS 15.5, macOS 12.3, tvOS 15.4, watchOS 8.5
    Android SDK:
      API Levels: 25, 29, 30, 31, 32, 33
      Build Tools: 29.0.2, 30.0.2, 30.0.3, 31.0.0, 33.0.0
      System Images: android-31 | Android TV ARM 64 v8a, android-31 | Android TV Intel x86 Atom, android-31 | Google TV ARM 64 v8a, android-31 | Google TV Intel x86 Atom, android-31 | Google APIs ARM 64 v8a, android-31 | Google APIs Intel x86 Atom_64, android-31 | Google Play ARM 64 v8a, android-31 | Google Play Intel x86 Atom_64, android-33 | Google APIs ARM 64 v8a
      Android NDK: Not Found
  IDEs:
    Android Studio: 2021.2 AI-212.5712.43.2112.8815526
    Xcode: 13.4.1/13F100 - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.16 - /Users/***/.jenv/shims/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.1.0 => 18.1.0 
    react-native: 0.70.1 => 0.70.1 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### What happened instead?

I try to build the android project through "npx react-native run-android" but I'm unable to build the project. I already ran ./gradlew clean, but still got the same issue.

```
"react": "18.1.0",
"react-native": "0.70.1",
"react-native-reanimated": "^2.10.0",
"react-native-vision-camera": "^2.14.1"
```

### Relevant log output

```shell
FAILURE: Build completed with 6 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:downloadGlog'.
> A failure occurred while executing de.undercouch.gradle.tasks.download.internal.DefaultWorkerExecutorHelper$DefaultWorkAction
   > java.net.SocketTimeoutException: Read timed out

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
java.lang.StackOverflowError (no error message)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

3: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:downloadFolly'.
> A failure occurred while executing de.undercouch.gradle.tasks.download.internal.DefaultWorkerExecutorHelper$DefaultWorkAction
   > java.net.SocketTimeoutException: Read timed out

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

4: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:downloadDoubleConversion'.
> A failure occurred while executing de.undercouch.gradle.tasks.download.internal.DefaultWorkerExecutorHelper$DefaultWorkAction
   > de.undercouch.gradle.tasks.download.org.apache.hc.core5.http.ConnectionClosedException: Premature end of Content-Length delimited message body (expected: 7012573; received: 702198)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

5: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-reanimated:downloadFolly'.
> A failure occurred while executing de.undercouch.gradle.tasks.download.internal.DefaultWorkerExecutorHelper$DefaultWorkAction
   > de.undercouch.gradle.tasks.download.org.apache.hc.core5.http.ConnectionClosedException: Premature end of Content-Length delimited message body (expected: 3465367; received: 601886)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

6: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-reanimated:downloadDoubleConversion'.
> A failure occurred while executing de.undercouch.gradle.tasks.download.internal.DefaultWorkerExecutorHelper$DefaultWorkAction
   > de.undercouch.gradle.tasks.download.org.apache.hc.core5.http.ConnectionClosedException: Premature end of Content-Length delimited message body (expected: 7012573; received: 4446756)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org
```

### Device

Android

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

lichmac, 2022-09-23:
Got more issue

`/node_modules/react-native/sdks/hermes/public -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -fno-limit-debug-info  -fPIC -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o -c ../../../../src/main/cpp/FrameProcessorRuntimeManager.cpp
  In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.cpp:5:
  In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:14:
  In file included from ../../../../src/main/cpp/reanimated-headers/AndroidScheduler.h:8:
  In file included from /Users/**/Documents/ReactNative/**/**/node_modules/react-native/ReactAndroid/src/main/jni/react/jni/CxxModuleWrapper.h:10:
  In file included from /Users/**/Documents/ReactNative/**/**/node_modules/react-native/ReactAndroid/src/main/jni/react/jni/CxxModuleWrapperBase.h:13:
  In file included from /Users/**/Documents/ReactNative/**/**/node_modules/react-native/ReactCommon/cxxreact/CxxModule.h:15:
  ../../../../build/third-party-ndk/folly/folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
  #include <boost/operators.hpp>
           ^~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [8/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [9/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o
  ninja: build stopped: subcommand failed.`

Neosoulink, 2022-12-20:
@lichmac, did you solve your issue or have an Idea of how to solve it?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy