margelo/react-native-vision-camera #1254: How can i get current camera's view/height❓

ikuruzum, 2022-09-23:
### Question

I'm making a barcode mask for my barcode scanner, but boundingBox prop is giving coordinates for camera's width and height, if i can get camera's width and height i will make something that will set the size in the right proportion so how can i get camera's width and height?

### What I tried

i tried to fix it by assigning a preset

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bglgwyng, 2022-09-24:
I think that you should use the `Frame` object's `width` and `height`.
I haven't seen the corresponding values from the device info.

mrousavy, 2022-10-06:
Format has a `videoWidth` and `videoHeight` prop

closed 2022-10-06 by mrousavy