margelo/react-native-vision-camera #1260: 🐛

DevelopCarlosWM, 2022-09-28:
### What were you trying to do?

I'm trying to run with RN 0.69.4 but get a task error

### What happened instead?

`FAILURE: Build failed with an exception.

* Where:
Build file 'node_modules/react-native-vision-camera/android/build.gradle' line: 588

* What went wrong:
Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
> Cannot expand ZIP '/node_modules/react-native-reanimated/android/react-native-reanimated-69-jsc.aar' as it does not exist.`

### Device

Android build

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jamninetyfive, 2022-11-30:
https://github.com/mrousavy/react-native-vision-camera/issues/1210#issuecomment-1331689881

jRichardCotting, 2023-06-19:
It seems to me that in your project the version of the react-native-reanimated package that you have installed doesn't includes the '**react-native-reanimated-69-jsc.aar'** file. 

You can check if your version has the file in the website: https://unpkg.com/browse/react-native-reanimated@2.4.1/android/

If you are using versions higher than the v2 of React reanimated, consider downgrading as react-native-vision-camera still doesn't support v3.  If you are already using some version of v2 in your react-native-reanimated check if the package exists in this bundle. 

In my case I was working with the version 2.1.0,  which doesn't includes the **"-jsc.arr"** files. Upgrading to v2.4.1 solved my problem.

2Senn, 2023-07-20:
Still facing this issue even after downgrading react native reanimated to v2. I used reanimated@2.4.1 and it fails to find the .ZIP file that ends with 72. i am assuming this refers to the react native version? It is a bit confusing, I am not sure why it requests that to begin with.

clonalejandro, 2023-07-22:
same issue

jacobjones123, 2023-08-01:
same issue here
"node_modules/react-native-reanimated/android/react-native-reanimated-72-hermes.aar as it does not exist."

        "react": "18.2.0",
        "react-native": "0.72.3",
        "react-native-reanimated": "^2.14.4",
        "react-native-vision-camera": "^2.15.5",

elibroftw, 2023-08-02:
How do you build the file ourselves? I would rather fork and make a build for react-native-72

JulesPatry, 2023-08-06:
same here as well

Silvman, 2023-08-08:
I ran into the same issue in new project with latest RN 0.72 + react-native-vision-camera

react-native-vision-camera v2 requires react-native-reanimated v2.x.x which does not support new RN 0.72
support for 0.72 is only available since RNR v3.3.0 (https://github.com/software-mansion/react-native-reanimated/issues/4668#issuecomment-1618948806), but you can't use RNR3 with react-native-vision-camera

ended up downgrading react-native to 0.71.12

VivekCEpic, 2023-08-08:
Same here as well

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy