margelo/react-native-vision-camera #1262: how to return Frame from frameprocessor and display it on screen

zheniagnedchik, 2022-09-29:
### Question

how to return Frame from frameprocessor and display it on screen

### What I tried

how to return Frame from frameprocessor and display it on screen

### VisionCamera Version

2.1.14

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

thomas-coldwell, 2022-10-01:
This will be enabled by #1198 soon - just need to add Android support!

zheniagnedchik, 2022-10-04:
> This will be enabled by #1198 soon - just need to add Android support!

For ios, the functionality in this pr is already working

zheniagnedchik, 2022-10-05:
> Yes you can use it already for iOS with the `useSyncFrameProcessor` hook, but please be aware this might change. Revisiting the PR shortly as I got busy with work, but will be working on Android support next week. Cheers!

Thank you very much for the detailed answer) can you please tell me if it is possible to take a photo with the filter applied?

xulihang, 2022-10-11:
I convert the frame to Bitmap/UIImage and then convert it to base64: https://github.com/tony-xlh/vision-camera-dynamsoft-label-recognizer/blob/328624d0d1daf3583886d2958a938ac90963bfdc/ios/VisionCameraDLRPlugin.swift#L88. Don't know if there are better options.

thomas-coldwell, 2022-10-22:
> Thank you very much for the detailed answer) can you please tell me if it is possible to take a photo with the filter applied?

Hey @zheniagnedchik - its not currently implemented, but support will definitely be added :)

thomas-coldwell, 2022-10-22:
Going to also mark this as a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/724. We are actively working on adding support for write-back frame processors in Vision Camera V3 (the in-progress PR for this functionality can be found at https://github.com/mrousavy/react-native-vision-camera/pull/1198).

closed 2022-10-22 by thomas-coldwell