margelo/react-native-vision-camera #1268: 🐛 Frame Processor throw an error: Can't find variable __scanCodes

SahandMahdavi, 2022-10-05:
### What were you trying to do?

I have updated my XCode and after that, I am facing this issue and the camera can't scan 

I also have the globals: ['__scanCodes'], inside my babel.config.json

### What happened instead?

The error occurs and the camera doesn't scan the QRCode

### Device

iphone 6 and iphone 13 pro

### VisionCamera Version

^2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-10-06:
Can you share the native logs on startup?

yuki-tz, 2022-10-07:
I think this is the same issue as https://github.com/mrousavy/react-native-vision-camera/issues/1244

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy