margelo/react-native-vision-camera #1270: 🐛 Cannot access file created by takePhoto through FileSystem

mortenmo, 2022-10-05:
### What were you trying to do?

Expo 46, iOS, 

The picture works fine and <Image> can use it, however I cannot access it in any other way.

But doing anything to this file file FileSystem operations (beyond `getInfo`) fails with a "<file> isn't readable." error.

`[Error: File '/private/var/mobile/Containers/Data/Application/D606F8F1-0B4C-41D7-81A8-46AF72BED567/tmp/ReactNative/8CED5BE1-61D5-43B7- ... FF9.jpeg' isn't readable.]`

This comment mentions the same issue on an enhancement to be able to select the directory images are stored in. 
https://github.com/mrousavy/react-native-vision-camera/issues/893#issuecomment-1221305774

Saving the image to image library does however work using `await MediaLibrary.saveToLibraryAsync(path)`, but permissions there are probably different.

### Reproduceable Code

```tsx
const { path } = await camera.current.takePhoto({
     qualityPrioritization: 'balanced',
});

const file = FileSystem.documentDirectory +
   path.substring(path.lastIndexOf('/') + 1);

await FileSystem.copyAsync({
   from: path,
   to: file,
});
```

### What happened instead?

Besides showing the image in an Image tag, there

### Relevant log output

```shell
[Error: File '/private/var/mobile/Containers/Data/Application/D606F8F1-0B4C-41D7-81A8-46AF72BED567/tmp/ReactNative/8CED5BE1-61D5-43B7- ... FF9.jpeg' isn't readable.]
```

### Device

iPhone 13 

### VisionCamera Version

2.14.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mortenmo, 2022-10-05:
MlKit also failed to access the image:

`import ImageLabeling from '@react-native-ml-kit/image-labeling';`

```
    ImageLabeling.label(path);   // Fails with image is nil which probably means no access
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

Try the `file://` prefix

closed 2023-09-30 by mrousavy

joshua-revolutions, 2023-10-09:
I got the same issue when sending the photo into Google mlkit. Adding file:// prefix did fix it for me.

mrousavy, 2023-10-09:
> Adding file:// prefix did fix it for me

This is not a bug, this is how the filesystem works. It's even in the docs that you need to add this if you want to read the file.