margelo/react-native-vision-camera #1272: 🐛 I'll be getting crash issues when using video capture

bhavnaproses, 2022-10-07:
### What were you trying to do?

 <Camera
                        onInitialized={() => {
                         setactive(true)
                        }}
                        onError={onError}
                        ref={camera}
                        style={styles.absoluteFill}
                        device={device}
                        isActive={isActive}
                        photo={true}
                        video={isVideo ? true : false}      
                      
                       audio={isVideo ? true : false}
                    />

### Reproduceable Code

```tsx
capture photo is working finally but video not working
```

### What happened instead?

app crashed

### Relevant log output

```shell
no any getting error
```

### Device

Vivi V23e 5g

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sylvobravo, 2022-10-07:
I'm having the same issue in my project

jaypateljetbro, 2022-12-21:
i'm facing same issue.

YashGupta61969, 2023-03-16:
Same here. App crashes when i set video={true'}

laurencetroyv, 2023-08-10:
facing the same problem with my Realme 8i

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy