margelo/react-native-vision-camera #1273: How to switch camera devices❓

seriousStar, 2022-10-08:
### Question

How to switch camera devices?

### What I tried

The doc says there are multi lens such as `wide-angle-camera`, `telephoto-camera`, `ultra-wide-angle-camera`. I would like to switch camera lens on camera view. I tried to find some solutions on doc, but not found yet to switch camera lens. I can select the initial camera as Doc says `const devices = useCameraDevices('wide-angle-camera')`, but I couldn't figure out how to switch camera lens once it's mounted.

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

demsr, 2022-12-01:
Yeah the docs are quite silent about this. But you can just use some state and change the "useCameraDevices"-hook like so:

```
const [deviceType, setDeviceType] useState(null); // use default lense at startup
const [deviceDir, setDeviceDir] useState("back");

const devices = useCameraDevices(deviceType);

const device = devices[deviceDir];

...

return (
<View>
  <Camera device={device} isActive={true} ... />
  
  <button  onPress={() => setDevieType("wide-angle-camera") />
  <button  onPress={() => setDevieType("telephoto-camera") />
  <button  onPress={() => setDevieType("ultra-wide-angle-camera") />
</View>
)
```

valn1, 2022-12-19:
@demsr but is there a way to switch between the devices without stopping recording?

demsr, 2022-12-21:
@valn1 are we talking about Back/Front camera or the different kind of lenses on the back? The latter should not be necessary, my camera switches between lenses depending on the zoom level.

valn1, 2023-04-10:
@demsr yeah, i was talking about switching between front and back. sorry for the late response

maxximee, 2023-06-27:
> Yeah the docs are quite silent about this. But you can just use some state and change the "useCameraDevices"-hook like so:
> 
> ```
> const [deviceType, setDeviceType] = useState(null); // use default lense at startup
> const [deviceDir, setDeviceDir] = useState("back");
> 
> 
> const devices = useCameraDevices(deviceType);
> 
> const device = devices[deviceDir];
> 
> ...
> 
> return (
> <View>
>   <Camera device={device} isActive={true} ... />
>   
>   <button  onPress={() => setDeviceType("wide-angle-camera") />
>   <button  onPress={() => setDeviceType("telephoto-camera") />
>   <button  onPress={() => setDeviceType("ultra-wide-angle-camera") />
> </View>
> )
> ```
> 
> //fixed some typos

Apreciate the code, to be correct you should close the "}" and button should be Button

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

hiteshawasthi11, 2024-06-18:
You can follow this code :

  const devices = Camera.getAvailableCameraDevices();
  const [device, setDevice] = useState(getCameraDevice(devices, 'back'));
  
    const handleFlipCamera = () => {
    const position = device?.position == 'front' ? 'back' : 'front';
    setDevice(getCameraDevice(devices, position));
  };

//use
     <Camera
            isActive={isCameraActive}
            device={device}
            codeScanner={codeScanner}
            style={{height: '100%'}}
            torch={torchEnabled ? 'on' : 'off'} 
          />