margelo/react-native-vision-camera #1276: Add text while recording

meetkkvelu, 2022-10-11:
### What feature or enhancement are you suggesting?

Dear team,
   I need a one new feature(Add custom text while recording). For example please check the video in the below link.(https://www.canva.com/design/DAFNpQo3f0o/079fPmij7AJZyP_Ak2AsYA/watch?utm_content=DAFNpQo3f0o&&utm_campaign=designshare&&utm_medium=link&&utm_source=publishsharelink).

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

https://www.canva.com/design/DAFNpQo3f0o/079fPmij7AJZyP_Ak2AsYA/watch?utm_content=DAFNpQo3f0o&&utm_campaign=designshare&&utm_medium=link&&utm_source=publishsharelink

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

thomas-coldwell, 2022-10-22:
Marking this as a duplicate of #724. We are actively working on adding support for write-back frame processors in Vision Camera V3 (the in-progress PR for this functionality can be found at #1198).

closed 2022-10-22 by thomas-coldwell

rajyuva1000, 2023-09-15:
is this feature available in visioncamera?

tomerh2001, 2024-02-03:
Is this available?