margelo/react-native-vision-camera #1279: 🐛 App crahsed when try to use frameProcessor for QR code scanning

ApexcupDeveloper, 2022-10-12:
### What were you trying to do?

 "react": "18.1.0",
 "react-native": "0.70.0",
 "react-native-reanimated": "^2.10.0",
"react-native-vision-camera": "^2.14.1",
"vision-camera-code-scanner": "^0.2.0",

### What happened instead?

The app crashed without any error message.

### Device

iPhone 11 Pro, iPhone 8s

### VisionCamera Version

^2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

CaioQuirinoMedeiros, 2022-10-13:
@ApexcupDeveloper I haven't tested it on iOS yet, but I'm having crash on Android, you too?
RN 0.69.6 here

ApexcupDeveloper, 2022-10-13:
Yes.

mrousavy, 2022-10-14:
No logs from Xcode attached. I cannot reproduce and there is no info here.

closed 2022-10-14 by mrousavy

ApexcupDeveloper, 2022-10-14:
App crashed without any information.

mrousavy, 2022-10-14:
The information is in Xcode. Check your Xcode logs.