margelo/react-native-vision-camera #1283: 🐛 Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'

scaralfred, 2022-10-16:
### What were you trying to do?

Hello, I get this error when running the android build.

"react": "18.1.0",
"react-native": "0.70.3",
"react-native-reanimated": "^2.11.0"

Steps I made:

rm node_modules
npm install
cd android ./gradlew clean cd ..
npm react-native run-android

### What happened instead?

Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.

### Relevant log output

```shell
> Build command failed.
  Error while executing process /Users/admin/Library/Android/sdk/cmake/3.18.1/bin/ninja with arguments {-C /Users/admin/Desktop/AwesomeProject/node_modules/react-native-vision-camera/android/.cxx/Debug/42l6f1n6/arm64-v8a VisionCamera}
  ninja: Entering directory `/Users/admin/Desktop/AwesomeProject/node_modules/react-native-vision-camera/android/.cxx/Debug/42l6f1n6/arm64-v8a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o 
  /Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --gcc-toolchain=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/React -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/React/Base -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactAndroid/src/main/jni -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon/callinvoker -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon/jsi -I/Users/admin/Desktop/AwesomeProject/node_modules/hermes-engine/android/include -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/sdks/hermes/API -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/sdks/hermes/public -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -c ../../../../src/main/cpp/VisionCameraScheduler.cpp
  In file included from ../../../../src/main/cpp/VisionCameraScheduler.cpp:5:
  ../../../../src/main/cpp/VisionCameraScheduler.h:7:10: fatal error: 'Scheduler.h' file not found
  #include "Scheduler.h"
           ^~~~~~~~~~~~~
  1 error generated.
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o 
  /Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --gcc-toolchain=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/React -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/React/Base -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactAndroid/src/main/jni -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon/callinvoker -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ReactCommon/jsi -I/Users/admin/Desktop/AwesomeProject/node_modules/hermes-engine/android/include -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/sdks/hermes/API -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/sdks/hermes/public -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
  In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
  ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found
  #include "RuntimeManager.h"
           ^~~~~~~~~~~~~~~~~~
  1 error generated.
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o 
  /Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --gcc-toolchain=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/admin/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/React -I/Users/admin/Desktop/AwesomeProject/node_modules/react-native/ [... cut ...]
Samsung S22

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

scaralfred, 2022-10-16:
Fixed downgrading react-native-reanimated from 2.11.0 to 2.10.0

JamesGs1994, 2022-10-16:
already my version is 2.10.0 react-native-reanimated

JamesGs1994, 2022-10-16:
My error is below

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: debug|arm64-v8a :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|arm64-v8a :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|arm64-v8a :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|arm64-v8a :  code such as
C/C++: debug|arm64-v8a :    project(ProjectName)
C/C++: debug|arm64-v8a :  near the top of the file, but after cmake_minimum_required().
C/C++: debug|arm64-v8a :  CMake is pretending there is a "project(Project)" command on the first
C/C++: debug|arm64-v8a :  line.
C/C++: debug|arm64-v8a :This warning is for project developers.  Use -Wno-dev to suppress it.
C/C++: debug|arm64-v8a :CMake Warning at CMakeLists.txt:178 (message):
C/C++: debug|arm64-v8a :  VisionCamera linking: FOR_HERMES=true

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a]
C/C++: ninja: Entering directory `D:\Ariveguru\fh-apps\tbotapp\node_modules\react-native-vision-camera\android\.cxx\Debug\6y2c5x5i\arm64-v8a'
C/C++: In file included from ../../../../src/main/cpp/VisionCameraScheduler.cpp:5:
C/C++: ../../../../src/main/cpp/VisionCameraScheduler.h:7:10: fatal error: 'Scheduler.h' file not found
C/C++: In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
C/C++: ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found
C/C++: In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.cpp:5:
C/C++: ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5.1/userguide/command_line_interface.html#sec:command_line_warnings
280 actionable tasks: 8 executed, 272 up-to-date

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> Build command failed.
  Error while executing process C:\Users\james\AppData\Local\Android\Sdk\cmake\3.18.1\bin\ninja.exe with arguments {-C D:\Ariveguru\fh-apps\tbotapp\node_modules\react-native-vision-camera\android\.cxx\Debug\6y2c5x5i\arm64-v8a VisionCamera}
  ninja: Entering directory `D:\Ariveguru\fh-apps\tbotapp\node_modules\react-native-vision-camera\android\.cxx\Debug\6y2c5x5i\arm64-v8a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  C:\Users\james\AppData\Local\Android\Sdk\ndk\21.4.7075529\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=aarch64-none-linux-android21 --gcc-toolchain=C:/Users/james/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/james/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/React -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/React/Base -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/ReactAndroid/src/main/jni -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/ReactCommon -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/ReactCommon/callinvoker -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/ReactCommon/jsi -ID:/Ariveguru/fh-apps/tbotapp/node_modules/hermes-engine/android/include -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/Tools -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/Registries -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/sdks/hermes/API -ID:/Ariveguru/fh-apps/tbotapp/node_modules/react-native/sdks/hermes/public -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -MF CMakeFiles\VisionCamera.dir\src\main\cpp\VisionCameraScheduler.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o -c ../../../../src/main/cpp/VisionCam [... cut ...]
    at async runOnAllDevices (D:\Ariveguru\fh-apps\tbotapp\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:109:5)
    at async Command.handleAction (D:\Ariveguru\fh-apps\tbotapp\node_modules\@react-native-community\cli\build\index.js:142:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

D:\Ariveguru\fh-apps\tbotapp>

spaceod, 2022-10-17:
> Fixed downgrading react-native-reanimated from 2.11.0 to 2.10.0

"react-native-reanimated": "^2.11.0" I have revert reanimated version from 2.11.0 to 2.10.0 but still getting the same error.

*Also tried below steps 👇 after changing module version but it didn't worked out. 
rm node_modules
npm install
cd android ./gradlew clean cd ..
npm react-native run-android

daba-faith, 2022-10-17:
Same issue, has this been fixed?

MirUmair, 2022-10-17:
Same issue, has this been fixed?

Jaybee4real, 2022-10-17:
@GeandersonTodimo did you try the suggestion above?

ArturoTorresMartinez, 2022-10-17:
Thank you guys so much for posting this issue, I just ran into it on a brand new react-native (0.70.3) project with nothing else installed besides this and it's dependencies.

@Jaybee4real I'll give your comment a try in a sec and I'll confirm if it works for anyone else having this issue

Thanks again, let's keep us posted about any updates on this thread!

GeandersonTodimo, 2022-10-17:
> Me too had the same issue.
> 
> ## Gradle 7.5.1
> Build time: 2022-08-05 21:17:56 UTC Revision: d1daa0cbf1a0103000b71484e1dbfe096e095918
> 
> Kotlin: 1.6.21 Groovy: 3.0.10 Ant: Apache Ant(TM) version 1.10.11 compiled on July 10 2021 JVM: 11.0.12 (Oracle Corporation 11.0.12+8-LTS-237) OS: Windows 10 10.0 amd64
> 
> "react": "18.1.0", "react-native": "0.70.3", "react-native-hole-view": "^2.1.2", "react-native-reanimated": "2.10.0", "react-native-svg": "^13.4.0", "react-native-vision-camera": "^2.14.1", "vision-camera-code-scanner": "^0.2.0"
> 
> I had ran the follows commands inside folder android: 1º ./gradlew clean 2º ./gradlew.bat installDebug --warning-mode all 3º cd .. npx react-native run-android.
> 
> But it delayed so mush time and I always have to do the same because occur the problem frequently.

I closed and after install more some libs and came back the problem.

    "axios": "^1.1.3",
    "date-fns": "^2.29.3",
    "react-native-config": "^1.4.11",
    "react-native-fs": "^2.20.0",
    "react-native-modalize": "^2.1.1",
    "react-native-paper": "^4.12.5",
    "react-native-vector-icons": "^9.2.0",

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> Build command failed.
  Error while executing process C:\Users\geanderson.silva\AppData\Local\Android\Sdk\cmake\3.18.1\bin\ninja.exe with arguments {-C C:\Users\geanderson.silva\Documents\WorkSpace\AppTodimo\SIATDMobile\node_modules\react-native-vision-camera\android\.cxx\Debug\363i6o1q\arm64-v8a VisionCamera}
  ninja: Entering directory `C:\Users\geanderson.silva\Documents\WorkSpace\AppTodimo\SIATDMobile\node_modules\react-native-vision-camera\android\.cxx\Debug\363i6o1q\arm64-v8a'  
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  C:\Users\geanderson.silva\AppData\Local\Android\Sdk\ndk\21.4.7075529\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=aarch64-none-linux-android21 --gcc-toolchain=C:/Users/geanderson.silva/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/geanderson.silva/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/React -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/React/Base -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/ReactAndroid/src/main/jni -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/ReactCommon -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/ReactCommon/callinvoker -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/ReactCommon/jsi -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/hermes-engine/android/include -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/Tools -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/Registries -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_modules/react-native/sdks/hermes/API -IC:/Users/geanderson.silva/Documents/WorkSpace/AppTodimo/SIATDMobile/node_mod [... cut ...]
    at C:\Users\geanderson.silva\Documents\WorkSpace\AppTodimo\SIATDMobile\node_modules\execa\index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (C:\Users\geanderson.silva\Documents\WorkSpace\AppTodimo\SIATDMobile\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:109:5)
    at async Command.handleAction (C:\Users\geanderson.silva\Documents\WorkSpace\AppTodimo\SIATDMobile\node_modules\@react-native-community\cli\build\index.js:142:9)
info Run CLI with --verbose flag for more details.

ArturoTorresMartinez, 2022-10-17:
> Fixed downgrading react-native-reanimated from 2.11.0 to 2.10.0

Thanks for the suggestion, unfortunately it does not solve my issue with the project I described above

m-maaz-azhar, 2022-10-17:
I am also facing the same issue, has this been fixed?

ArturoTorresMartinez, 2022-10-17:
> I am also facing the same issue, has this been fixed?

I am afraid not yet, I'm keeping this thread posted in case I find a solution

GeandersonTodimo, 2022-10-17:
> 

@ArturoTorresMartinez in my project if I remove these lib, exemplo below came back working. But I also need them to scann barr codes and i would like to working as others libs without problems during build.

yarn remove react-native-vision-camera
yarn remove vision-camera-code-scanner react-native-hole-view

juanpablomorenomartin, 2022-10-17:
I suppose that this library will work for some specific version and that it will break with some update, currently I created a new project and I have this same problem, can someone advise me with which version this works

irodeanu, 2022-10-18:
> > > 
> > 
> > 
> > @ArturoTorresMartinez in my project if I remove these lib, exemplo below came back working. But I also need them to scann barr codes and i would like to working as others libs without problems during build.
> > yarn remove react-native-vision-camera yarn remove vision-camera-code-scanner react-native-hole-view
> 
> @GeandersonTodimo Indeed you are right, that seems to be the root of the issue
> 
> One not-so-great alternative is to fall back to the package that sent most of us here (react-native-camera) It sure needs a few tweaks with patch-package but if anyone is in a rush to get a scanner working I believe that's one way to go about it :/

I think is the best solution.

ikuruzum, 2022-10-18:
> @ArturoTorresMartinez in my project if I remove these lib, exemplo below came back working. But I also need them to scann barr codes and i would like to working as others libs without problems [...]

i can say it's not about react-native-hole-view, i don't use that but i get same error

wasiquehaider, 2022-10-18:
I fixed it by following the below steps :)

```
- watchman watch-del-all 
- yarn cache clean --force
-  npm cache clean --force # optional - in case you have used npm
-  rm -rf node_modules/ rm yarn.lock
-  rm package-lock.json # optional - in case you have used npm 
- rm -rf $TMPDIR/metro-* 
- rm -rf $TMPDIR/haste-*
-  yarn install / npm install
-  yarn start --reset-cache
- cd ios && rm -rf Podfile.lock && rm -rf pods && pod install --verbose
```

thuantvDr4, 2022-10-18:
yeah! i confirm this solution, because I faced the issue and i fixed by the solution

you should use 
"react-native-reanimated": "2.10.0", 
for
"react-native-vision-camera": "2.14.1",

remember, remove node_modules and  gradlew clean, and then run 
yarn android

wish successful for you !

ikuruzum, 2022-10-18:
> I fixed it by following the below steps :)
>
> ```
> - watchman watch-del-all
> - yarn cache clean --force
> -  npm cache clean --force # optional - in case you have used npm
> -  rm -rf node_modules/ rm [...]

that fixed my issue too

ikuruzum, 2022-10-18:
oh wait, it didn't

jwoodmansey, 2022-10-18:
We were seeing this in our `"react-native": "0.70.3"` spike, but all is good on `0.69.6`, so if you can try downgrading.

ikuruzum, 2022-10-18:
i tried react native 0.68.2, 0.70.3 and 0.69.5, i assume it's not about react-native version

jwoodmansey, 2022-10-18:
> i tried react native 0.68.2, 0.70.3 and 0.69.5, i assume it's not about react-native version

Damn, I will try work out if we changed anything else.

ikuruzum, 2022-10-18:
> > Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: debug|arm64-v8a :CMake Warning (dev) in CMakeLists.txt:
C/C++: debug|arm64-v8a :  No project() command is present.  The top-level CMakeLists.txt file must
C/C++: debug|arm64-v8a :  contain a literal, direct call to the project() command.  Add a line of
C/C++: debug|arm64-v8a :  code such as
C/C++: debug|arm64-v8a :    project(ProjectName)
C/C++: debug|arm64-v8a :  near the top of the file, but after cmake_minimum_required().
C/C++: debug|arm64-v8a :  CMake is pretending there is a "project(Project)" command on the first
C/C++: debug|arm64-v8a :  line.
C/C++: debug|arm64-v8a :This warning is for project developers.  Use -Wno-dev to suppress it.
C/C++: debug|arm64-v8a :CMake Warning at CMakeLists.txt:178 (message):
C/C++: debug|arm64-v8a :  VisionCamera linking: FOR_HERMES=true

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a]
C/C++: ninja: Entering directory `/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-vision-camera/android/.cxx/Debug/20233w2g/arm64-v8a'
C/C++: In file included from ../../../../src/main/cpp/VisionCameraScheduler.cpp:5:
C/C++: ../../../../src/main/cpp/VisionCameraScheduler.h:7:10: fatal error: 'Scheduler.h' file not found
C/C++: In file included from ../../../../src/main/cpp/VisionCamera.cpp:3:
C/C++: ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found
C/C++: In file included from ../../../../src/main/cpp/FrameProcessorRuntimeManager.cpp:5:
C/C++: ../../../../src/main/cpp/FrameProcessorRuntimeManager.h:13:10: fatal error: 'RuntimeManager.h' file not found

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5.1/userguide/command_line_interface.html#sec:command_line_warnings
302 actionable tasks: 297 executed, 5 up-to-date
Note: /Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-sound/android/src/main/java/com/zmxv/RNSound/RNSoundModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-wheel-pick/android/src/main/java/com/tron/ReactWheelCurvedPicker.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-wheel-pick/android/src/main/java/com/tron/ReactWheelCurvedPickerManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/android/app/src/debug/java/com/evobulutdepoyonetim/ReactNativeFlipper.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> Build command failed.
  Error while executing process /Users/ercanilik/Library/Android/sdk/cmake/3.18.1/bin/ninja with arguments {-C /Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-vision-camera/android/.cxx/Debug/20233w2g/arm64-v8a VisionCamera}
  ninja: Entering directory `/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native-vision-camera/android/.cxx/Debug/20233w2g/arm64-v8a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o 
  /Users/ercanilik/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --gcc-toolchain=/Users/ercanilik/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/ercanilik/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/React" -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/React/Base" -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/ReactAndroid/src/main/jni" -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni" -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/ReactCommon" -I"/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/react-native/ReactCommon/callinvoker" -I"/Users/ercanilik/Desktop/Son versiyon react na [... cut ...]
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/Users/ercanilik/Desktop/Son versiyon react native/EvoBulutDepoYonetim/node_modules/@react-native-community/cli/build/index.js:142:9)
    
    
    This is the error i encountered

Fari056, 2022-10-18:
cd android  => ./gradlew clean
delete node_module
"react-native-reanimated": "2.10.0",
for
"react-native-vision-camera": "2.14.1",
if  you have   '^'  this sign with versions remove it first
yarn / npm i

olegmilan, 2022-10-18:
> yeah! i confirm this solution, because I faced the issue and i fixed by use it
> 
> you should use "react-native-reanimated": "2.10.0", for "react-native-vision-camera": "2.14.1",
> 
> remember, remove node_modules and gradlew clean, and then run yarn android
> 
> wish successful for you !

Indeed, downgrading `react-native-reanimated` from `2.11.0` to `2.10.0` fixed issue for me. I got MacBook Pro with M1 Pro chip.
Thanks. dude

cc: @scaralfred

ArturoTorresMartinez, 2022-10-18:
> cd android => ./gradlew clean delete node_module "react-native-reanimated": "2.10.0", for "react-native-vision-camera": "2.14.1", if you have '^' this sign with versions remove it first yarn / npm i

Hello, I can also confirm this downgrade worked for me, however, anyone trying this fix should try to build an apk or bundle and double check for errors there

olegmilan, 2022-10-18:
that's right @ArturoTorresMartinez, build is generating for me, but app crashes at start

ArturoTorresMartinez, 2022-10-19:
> that's right @ArturoTorresMartinez, build is generating for me, but app crashes at start

What error are you getting?

thiagosimon, 2022-10-19:
> 
@Fari056  solution worked for me =]

"react": "^18.2.0",
"react-native": "^0.70.1",
"react-native-reanimated": "2.10.0",
"react-native-vision-camera": "2.14.1",

juanpablomorenomartin, 2022-10-19:
Nothing they say solves the problem, I don't think the solution has to be to use a deprecated library like react-native-camera

Namqv, 2022-10-19:
react-native-reanimated delete headers in react-native-reanimated/Common)/cpp/
But react-native-vision-camera using /react-native-reanimated/Common/cpp/headers/
So, an error occurred when the Scheduler.h and RuntimeManager.h files were missing
Fix : node_modules/react-native-vision-camera/android/CMakeLists.txt edit path

kamo9527, 2022-10-19:
> I fixed it by following the below steps :)
>
> ```
> - watchman watch-del-all
> - yarn cache clean --force
> -  npm cache clean --force # optional - in case you have used npm
> -  rm -rf node_modules/ rm [...]
@wasiquehaider  @thuantvDr4  @ikuruzum 
hi ,It  didn't work, the same error still .  which verson of react native ?

kamo9527, 2022-10-19:
> react-native-reanimated delete headers in react-native-reanimated/Common)/cpp/ But react-native-vision-camera using /react-native-reanimated/Common/cpp/headers/ So, an error occurred when the Scheduler.h and RuntimeManager.h files were missing Fix : node_modules/react-native-vision-camera/android/CMakeLists.txt edit path Or yarn "react-native-reanimated": "2.10.0",

@Namqv  please show more detail and the sets how you fix , thanks/

Namqv, 2022-10-19:
Option 1 :
delete node_modules
edit package.json
"react": "^18.2.0",
"react-native": "^0.70.1",
"react-native-reanimated": "2.10.0", #not using "^2.10.0"
"react-native-vision-camera": "^2.14.1",

watchman watch-del-all 
yarn cache clean --force
yarn install
////
check node_modules/Common/headers does it exist?
if exist next
not exist Op 2
///
cd android => ./gradlew clean
yarn android
Option 2 :
cd node_modules/react-native-vision-camera/android edit CMakeLists.txt
find
       "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/AnimatedSensor"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/Tools"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/SpecTools"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/SharedItems"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/Registries"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/headers/LayoutAnimations"
    
edit
"${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/AnimatedSensor"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/Tools"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/SpecTools"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/SharedItems"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/Registries"
        "${NODE_MODULES_DIR}/react-native-reanimated/Common/cpp/LayoutAnimations"

TensionCoding, 2022-10-19:
As was suggested above, downgrading to 2.10.0 without the "^" solved the problem for me

euclidesdry, 2022-10-19:
> Fixed downgrading react-native-reanimated from 2.11.0 to 2.10.0

the version of "react-native-reanimated" 2.10.0 solved my problem too

amochkin, 2022-10-20:
I've made a new PR that fixes this issue for newer version of `react-native-reanimated`, so it is up to maintainer now.

thiavila-sigma, 2022-10-20:
@olegmilan Are you building just for android or IOS too? And is Fabric enabled in your project?

kwangjaeyoo, 2022-10-21:
react-native-reanimated 2.15.0 
react-native-reanimated 2.11.0 
==> same issue

euclidesdry, 2022-10-21:
> react-native-reanimated 2.15.0 react-native-reanimated 2.11.0 ==> same issue

I'm using version 2.10.0 is much more stable.
Sometimes you'll need to run a few 2 or 3 times and clean the Gradle cache to work, but, it is more stable than others.

alfredarcifa, 2022-10-21:
> I've made a new PR that fixes this issue for newer version of `react-native-reanimated`, so it is up to maintainer now.

Hi!

I see your PR and I want to ask where, manually, I have to put your code.
In node_modules/react-native-reanimated/android/CMakeLists.txt ?

Thanks

kamo9527, 2022-10-23:
> react-native-reanimated delete headers in react-native-reanimated/Common)/cpp/ But react-native-vision-camera using /react-native-reanimated/Common/cpp/headers/ So, an error occurred when the [...]

@euclidesdry hi ,It didn't work, the same error still

kamo9527, 2022-10-23:
dont import 'react-native-reanimated'

ildfreelancer, 2022-10-23:
A Workaround for those who gets this issue,

Downgrade **react-native-reanimated 2.10.0**

wasiquehaider, 2022-10-24:
> @wasiquehaider @thuantvDr4 @ikuruzum hi ,It didn't work, the same error still . which verson of react native ?

 "react": "18.1.0",
 "react-native": "0.70.0",
 "react-native-vision-camera": "^2.14.1",
 "vision-camera-code-scanner": "^0.2.0"

ubugnu, 2022-10-28:
This is how I fixed it:

The problem was a missing directory called `boost` in `./node_modules/react-native-vision-camera/android/build/third-party-ndk/folly`, so I searched for the boost version in my project, with for example:

```bash
find . -name boost
```

Which gave:

```bash
./node_modules/react-native-reanimated/android/build/third-party-ndk/boost/boost_1_76_0/boost
# ...
```

Then on this page https://www.boost.org/users/history/ I downloaded the right version, for my case it was `1.76.0`, I decompressed the file, then I copied the directory `./boost` to `./node_modules/react-native-vision-camera/android/build/third-party-ndk/folly`.

Now it works.

david-r-frederick, 2022-10-31:
@ubugnu your fix worked for me. Downgrading react-native-reanimated to 2.10.0 did not work.

alfredarcifa, 2022-11-01:
> @ubugnu your fix worked for me. Downgrading react-native-reanimated to 2.10.0 did not work.

With new version there isn't this problem.
You can use last react-native-reanimated version

soso-15315, 2022-11-07:
We have confirmed that this problem does not occur in the latest version.

"react-native-vision-camera": "2.15.2", 
"react-native-reanimated": "2.12.0",

valn1, 2022-11-07:
> We have confirmed that this problem does not occur in the latest version.
> 
> "react-native-vision-camera": "2.15.2", "react-native-reanimated": "2.12.0",

instead, the app crashes before starting without throwing any errors

mkurczewski, 2022-11-07:
> > We have confirmed that this problem does not occur in the latest version.
> > "react-native-vision-camera": "2.15.2", "react-native-reanimated": "2.12.0",
> 
> instead, the app crashes before starting without throwing any errors

I also had to implement the solution from this post https://stackoverflow.com/a/74342523/3763750 and after that it started working again 🎉

ccubukcu, 2022-11-08:
The build issues are related to React Native and started on 4th of November after the release of the first RC of RN 0.71. They released hotfixes for React Native versions 63-70, see here: https://github.com/facebook/react-native/issues/35210

valn1, 2022-11-08:
works fine in debug mode, still crashes on startup in release mode, i'll conduct more tests to see if it actually has something to do with that, the problem is i have no info about it.
any help with debugging this would be much appreciated

amochkin, 2022-11-08:
> works fine in debug mode, still crashes on startup in release mode, i'll conduct more tests to see if it actually has something to do with that, the problem is i have no info about it.
> 
> any help with debugging this would be much appreciated

Is it iOS or Android? Simulator or real device? Did you reinstall node_modules? If iOS did you reintegrate the pods?

valn1, 2022-11-08:
Android, real devices, reinstalled node_modules
haven't tried ios yet

valn1, 2022-11-08:
here are some interesting logs i got from logcat

[image]

[image]

specially this one:
[image]

[image]

valn1, 2022-11-08:
[image]
fixed it

flaviocoppola, 2022-11-16:
> 
How can I solve the problem?

ameerarx, 2022-11-19:
> [image] fixed it

does it need to be changed in the lib itself or in the project?

[21 more replies]

closed 2023-09-01 by mrousavy