margelo/react-native-vision-camera #1284: Execution failed for task ':react-native-vision-camera:generateJsonModelDebug'. > java.lang.NullPointerException (no error message)🐛

Jaybee4real, 2022-10-16:
### What were you trying to do?

I was trying to build the app with the package on android

### Reproduceable Code

```tsx
Just try to run the package on android
```

### What happened instead?

The app crashes with the message above at 97% with the below error

### Relevant log output

```shell
> Task :react-native-vision-camera:generateJsonModelDebug FAILED
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings
831 actionable tasks: 831 executed
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/appcenter-crashes/android/src/main/java/com/microsoft/appcenter/reactnative/crashes/AppCenterReactNativeCrashesListener.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@invertase/react-native-apple-authentication/android/src/main/java/com/RNAppleAuthentication/AppleAuthenticationAndroidModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-async-storage/async-storage/android/src/main/java/com/reactnativecommunity/asyncstorage/AsyncStorageModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-async-storage/async-storage/android/src/main/java/com/reactnativecommunity/asyncstorage/AsyncStoragePackage.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/react-native-code-push/android/app/src/main/java/com/microsoft/codepush/react/CodePushNativeModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/react-native-code-push/android/app/src/main/java/com/microsoft/codepush/react/CodePushNativeModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-community/cameraroll/android/src/main/java/com/reactnativecommunity/cameraroll/CameraRollModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-firebase/analytics/android/src/main/java/io/invertase/firebase/analytics/UniversalFirebaseAnalyticsModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-firebase/dynamic-links/android/src/main/java/io/invertase/firebase/dynamiclinks/ReactNativeFirebaseDynamicLinksModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-firebase/messaging/android/src/main/java/io/invertase/firebase/messaging/ReactNativeFirebaseMessagingModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-google-signin/google-signin/android/src/main/java/com/reactnativegooglesignin/RNGoogleSigninModule.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-google-signin/google-signin/android/src/main/java/com/reactnativegooglesignin/RNGoogleSigninModule.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some input files use or override a deprecated API.
[... 121 lines cut ...]

    at makeError (/Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/execa/index.js:174:9)
    at /Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/Users/jaybee4real/Documents/Programming-codes/Daba/mobile-client/node_modules/@react-native-community/cli/build/index.js:192:9)
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

### Device

Samsung S10 plus (Android 12)

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jaybee4real, 2022-10-17:
Hey everyone, i think i found a solution to it, please try this and don't forget to come back to indicate if it works for you! 😊

- --> **Step 1**: Go into your gradle-wrapper.properties file and upgrade your gradle version to at least version 7.0.2 (See image below)

[image]

- ---> **Step 2**: Go into your android/build.gradle and add an extra classpath to your dependencies in android/build.gradle
  `classpath 'com.android.tools.build:gradle:7.0.0'`
 (See image below)

[image]

- ---> **Optional Step**: you might want to change all your dependencies URL to use "https" instead of "http" in your allProjects.repositories in android/build.gradle as you might face an error when building if you do not do this (See image below)

[image]

princefishthrower, 2022-10-18:
@Jaybee4real - I have two higher versions: gradle 7.2 and build:gradle:7.0.1 and I still can't build for android when this library installed.

Perhaps the combination of _only_ 7.0.2 and 7.0.0 work?

Jaybee4real, 2022-10-18:
Yes, that is what I found out @princefishthrower, you can try that

princefishthrower, 2022-10-18:
@Jaybee4real - I still can't build after implementing these changes. What version of `react-native-vision-camera` are you using?

Jaybee4real, 2022-10-18:
"react-native-vision-camera": "^2.14.1"

MohamedAhmed122, 2022-10-18:
> Hey everyone, i think i found a solution to it, please try this and don't forget to come back to indicate if it works for you! 😊
> 
> * --> **Step 1**: Go into your gradle-wrapper.properties file and upgrade your gradle version to at least version 7.0.2
>   (See example image below)
> 
> [image]
> 
> * ---> **Step 2**: Go into your android/build.gradle and add an extra classpath to your dependencies in android/build.gradle
>   `classpath 'com.android.tools.build:gradle:7.0.0'`
>   (See example image below)
> 
> [image]
> 
> * ---> **Optional Step**: you might want to change all your dependencies URL to use "https" instead of "http" in your allProjects.repositories in android/build.gradle as you might face an error when building if you do not do this
>   (See example image below)
> 
> [image]

@Jaybee4real  I  followed these steps, but I end up having this error, do you know how to fix it?

```
> Task :react-native-gradle-plugin:compileKotlin FAILED

FAILURE: Build failed with an exception.
```

Jaybee4real, 2022-10-18:
@MohamedAhmed122 

Check your bash_profile, do you see these values there?

`export ANDROID_SDK_ROOT=$HOME/Library/Android/sdk
export PATH=$PATH:$ANDROID_SDK_ROOT/emulator
export PATH=$PATH:$ANDROID_SDK_ROOT/platform-tools
**export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)**`

(Especially the JAVA_HOME)

[image]

Jaybee4real, 2023-01-12:
Will be closing this now

closed 2023-01-12 by Jaybee4real

akhilkrishnan1268, 2024-03-26:
> Hey everyone, i think i found a solution to it, please try this and don't forget to come back to indicate if it works for you! 😊
>
> * --> **Step 1**: Go into your gradle-wrapper.properties file and [...]

This worked for me.