margelo/react-native-vision-camera #1287: CLANG_CXX_LANGUAGE_STANDARD mismatch for iOS podspecs

amochkin, 2022-10-18:
### What were you trying to do?

package.json dependencies:
```
    "react-native": "0.70.3",
    "react-native-reanimated": "^2.11.0",
    "react-native-vision-camera": "^2.14.1"
```

empty project initialization fails on: `pod install`

### Reproduceable Code

```tsx
N/A
```

### What happened instead?

Pods installation fails.

Quite obviously, **VisionCamera.podspec.json** have:
```json
{
  "xcconfig": {
    "CLANG_CXX_LANGUAGE_STANDARD": "c++14",
    "HEADER_SEARCH_PATHS": "\"$(PODS_ROOT)/boost\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/glog\" \"$(PODS_ROOT)/RCT-Folly\" \"${PODS_ROOT}/Headers/Public/React-hermes\" \"${PODS_ROOT}/Headers/Public/hermes-engine\"",
    "OTHER_CFLAGS": "$(inherited) -DFOLLY_NO_CONFIG -DFOLLY_MOBILE=1 -DFOLLY_USE_LIBCPP=1 -DRNVERSION=70"
  }
}
```

While **hermes-engine.podspec.json** have:
```json
{
  "xcconfig": {
    "CLANG_CXX_LANGUAGE_STANDARD": "c++17",
    "CLANG_CXX_LIBRARY": "compiler-default",
    "GCC_PREPROCESSOR_DEFINITIONS": "HERMES_ENABLE_DEBUGGER=1"
  }
}
```

### Relevant log output

```shell
[!] Can't merge user_target_xcconfig for pod targets: ["VisionCamera", "hermes-engine"]. Singular build setting CLANG_CXX_LANGUAGE_STANDARD has different values.
```

### Device

N/A

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

amochkin, 2022-10-18:
And I don't see any reason why not just bumping **CLANG_CXX_LANGUAGE_STANDARD** to `c++17`: iOS stack uses Swift anyway...

Rustamyan94, 2022-10-21:
It's helped me.
run `npx pod-install` Instead, `cd ios && pod install cd ..`

amochkin, 2022-10-21:
@Rustamyan94 Ok, I will check

frankbolviken, 2022-10-24:
Same issue here

R8P, 2022-10-25:
Same issue here

yeminlideveloper, 2022-10-25:
Same issue here!!!

akanomi-44, 2022-10-25:
Same issue here for both android and ios xD

ombogdan, 2022-10-26:
same problem

andybui090, 2022-10-26:
The same

LazyAfternoons, 2022-10-27:
I have the same issue, bumping the value should do it.

amochkin, 2022-10-28:
I've PR'ed this fix as I found it is well working and totally fixing the problem with no side effects.

amochkin, 2022-10-28:
@mrousavy Hi, Marc, can you please review https://github.com/mrousavy/react-native-vision-camera/pull/1309 ?
Thanks.

Side note:
After applying this fix and problem persisting, it is required reinstall pods like so: 
```shell
pod deintegrate
pod install
```

closed 2022-10-31 by mrousavy

CoryWritesCode, 2023-01-25:
@mrousavy I was still having issues after #1309 so I dug in and found the fix was in the wrong place. Made #1432 to correct this.

PR #1309 (2022-10-31): Update CLANG_CXX_LANGUAGE_STANDARD to C++17
## What

Bumped CLANG_CXX_LANGUAGE_STANDARD to C++17 to fix #1287.

## Changes

Nothing is changed except version is bumped to C++17.

## Tested on

iPhone 12 Pro Max
iPhone Simulators

## Related issues

Fixes #1287