margelo/react-native-vision-camera #1290: Vision Camera:downloadBoost is very very slow on build time🐛

hmirzaa, 2022-10-19:
### What were you trying to do?

I have followed all the steps and for the first build Vision camera downloadBoost failed, Folly download failed on build time.

Then I started the build again, Configuration part was 100% but on Executing after 96% it started downlaod React-Native-Vision-Camera:DownloadBoost 124MB and its very slow. My internet is working perfectly but this download is like 350Kb in 10 minutes.

### Reproduceable Code

```tsx
Npm run android
```

### What happened instead?

Downloading was very slow. I stopped the build

### Device

OPPO Reno 4

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hmirzaa, 2022-10-19:
Tried to clean everything and build again. Now "react native vision camera boost download failed" on build timr

closed 2022-10-20 by hmirzaa

icc-romeu, 2022-11-01:
Hi, did you solve this finally? I think it might be related to server issues but... I'm stuck there.

allen-hsu, 2023-01-03:
hmmm, I got the same issue...

EwaGuziejko, 2023-01-10:
Issue still exists :/

punov, 2023-01-10:
Same, today it's incredibly slow again.

Hemuxue, 2023-04-23:
same issue。

farhan-spinny, 2023-09-11:
same issue

fimak, 2023-10-03:
same issue

sarthakpranesh, 2023-11-15:
same issue