margelo/react-native-vision-camera #1294: Has anyone been able to get Ml Kit Frame Processors this to work with react native 0.70 > ❓

bearsworth, 2022-10-20:
### Question

I have been able to install frame processors without issue. Everything works even the frame processor, but once I try to use the the image labeler like the example in frame processors, the app crashes instantly. I have tried all kinds of things to get this to work, changing versions of react reanimated, etc. But to no avail.

If someone does manage to get ML kit frame processors work, please share your React reanimated version and any other versioning you had to specify. It would greatly help. 

### What I tried

All kinds of stuff that I cannot fathom to write it. 

### VisionCamera Version

Multiple versions tried, nothing worked. 

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

KonstantinZhukovskij, 2022-10-25:
Maybe someone has more working solutions than removing the framework?

VladyslavMartynov, 2022-10-26:
Have the same error when was trying to use Frame Processors are not enabled in RN v 0.70.1. Any solutions?

bearsworth, 2022-10-26:
> Have the same error when was trying to use Frame Processors are not enabled in RN v 0.70.1. Any solutions?

I eventually gave up and decided to use expo camera. It might work for some purposes but you are limited to only barcode scanning + face detection. Pretty much the two things I actually needed fortunately. But this is not a great solution if you plan to use ML kit for other stuff. 

I believe vision camera has issues and you may need to download react native camera and figure it out or downgrade your version (which is quite sad).

adolfogc, 2022-10-28:
It works for me on Android for barcode scanning using RN 0.70.4; need to test it on iOS.

alexstanbury, 2022-12-23:
Yes, I have a fork of vision-camera-code-scanner (had to fork it due to an android mlkit dependency mismatch and it's not maintained) working and my own image labelling ml-kit frame processor for custom models working on iOS and 0.70.5.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy