margelo/react-native-vision-camera #1295: 🐛 libc++abi: terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

MeesMaas, 2022-10-20:
### What were you trying to do?

I created a new project using `npx react-native init iOSProject` and installed `react-native-vision-camera` and `react-native-reanimated`. This works just fine. But when I create a FrameProcessor using the guide from the [docs](https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors-plugins-ios) it crashes. 

### Reproduceable Code

```tsx
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 *
 * @format
 * @flow strict-local
 */

import React, {useState} from 'react';
import {StyleSheet, Text, View} from 'react-native';
import {runOnJS} from 'react-native-reanimated';

import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';

export function scanQrCode(frame) {
  'worklet';
  // eslint-disable-next-line no-undef
  return __scanQrCode(frame);
}

function App() {
  const devices = useCameraDevices();
  const device = devices.back;
  const [qrCode, setQrCode] = useState({});

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const scannedQrCode = scanQrCode(frame);

    // runOnJS(setQrCode)(scannedQrCode);

    // console.log(`QR Codes in Frame: ${JSON.stringify(scannedQrCode)}`);
  }, []);

  return (
    <View style={styles.flex}>
      <View style={styles.margin}>
        <Text>Following information was found:</Text>
        <Text>Type: {qrCode.type}</Text>
        <Text>Value: {qrCode.value}</Text>
      </View>
      {device && (
        <Camera
          device={device}
          isActive={true}
          style={styles.flex}
          frameProcessor={frameProcessor}
          frameProcessorFps={12}
        />
      )}
    </View>
  );
}

const styles = StyleSheet.create({
  flex: {
    flex: 1,
    backgroundColor: 'white',
  },
  margin: {
    marginVertical: 20,
    marginLeft: 10,
  },
});

export default App;
```

### What happened instead?

The crash:
```
libc++abi: terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
```

I tried the following:

- Different versions of react-native-reanimated
- Downgraded xCode to 13.4.1
- Tried Swift and Objective-C

### Device

iPhone 12 (iOS 16)

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

QSuraj, 2022-10-25:
@mrousavy Any idea why this might be happening?

mrousavy, 2022-10-25:
How should I if I don't have a source code for the scanQRCode

mrousavy, 2022-10-25:
Is this your FP Plugin?

```objc
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/Frame.h>

@interface QRCodeFrameProcessorPlugin : NSObject
@end

@implementation QRCodeFrameProcessorPlugin

static inline id scanQRCodes(Frame* frame, NSArray* args) {
  CMSampleBufferRef buffer = frame.buffer;
  UIImageOrientation orientation = frame.orientation;
  // code goes here
  return @[];
}

VISION_EXPORT_FRAME_PROCESSOR(scanQRCodes)

@end
```

closed 2022-10-25 by mrousavy

QSuraj, 2022-10-25:
> 

I'm receiving the crash even with FP plugin you shared 

```objc
libc++abi: terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
```

QSuraj, 2022-10-25:
Usage in code -

Reference to the native processor -
```js
const processFrame = ( frame ) => {
  'worklet';

  // eslint-disable-next-line no-undef
  return __processFrame( frame );
};
```

Using the native processor as so -
```js
const frameProcessor = useFrameProcessor( ( frame ) => {
    'worklet';

    const data = processFrame( frame );
    console.log( 'data', data );
    
  }, [] );
```

mateusz1913, 2022-10-27:
This is fixed with https://github.com/mateusz1913/vision-camera-plugin-builder/pull/24 and available with vision-camera-plugin-builder v0.2.0