margelo/react-native-vision-camera #1296: 🐛 Cannot swap between cameras while recording on Android

balloman, 2022-10-22:
### What were you trying to do?

When recording a video on iOS, I am able to swap the camera and continue the recording with no issue. This seems to not be possible on android, instead the recording fails when this happens. This feature is supported on apps like Tiktok and Snapchat, and is also supported by the native camera application.

- Note that the reproducible code is simply the demo application, modified to allow the user to press the swap button while recording, by setting the capture button to record instead of take a picture when tapped

### Reproduceable Code

```tsx
import React, { useCallback, useMemo, useRef } from 'react';
import { StyleSheet, View, ViewProps } from 'react-native';
import {
  PanGestureHandler,
  PanGestureHandlerGestureEvent,
  State,
  TapGestureHandler,
  TapGestureHandlerStateChangeEvent,
} from 'react-native-gesture-handler';
import Reanimated, {
  cancelAnimation,
  Easing,
  Extrapolate,
  interpolate,
  useAnimatedStyle,
  withSpring,
  withTiming,
  useAnimatedGestureHandler,
  useSharedValue,
  withRepeat,
} from 'react-native-reanimated';
import type { Camera, PhotoFile, TakePhotoOptions, TakeSnapshotOptions, VideoFile } from 'react-native-vision-camera';
import { CAPTURE_BUTTON_SIZE, SCREEN_HEIGHT, SCREEN_WIDTH } from './../Constants';

const PAN_GESTURE_HANDLER_FAIL_X = [-SCREEN_WIDTH, SCREEN_WIDTH];
const PAN_GESTURE_HANDLER_ACTIVE_Y = [-2, 2];

const START_RECORDING_DELAY = 200;
const BORDER_WIDTH = CAPTURE_BUTTON_SIZE * 0.1;

interface Props extends ViewProps {
  camera: React.RefObject<Camera>;
  onMediaCaptured: (media: PhotoFile | VideoFile, type: 'photo' | 'video') => void;

  minZoom: number;
  maxZoom: number;
  cameraZoom: Reanimated.SharedValue<number>;

  flash: 'off' | 'on';

  enabled: boolean;

  setIsPressingButton: (isPressingButton: boolean) => void;
}

const _CaptureButton: React.FC<Props> = ({
  camera,
  onMediaCaptured,
  minZoom,
  maxZoom,
  cameraZoom,
  flash,
  enabled,
  setIsPressingButton,
  style,
  ...props
}): React.ReactElement => {
  const pressDownDate = useRef<Date | undefined>(undefined);
  const isRecording = useRef(false);
  const recordingProgress = useSharedValue(0);
  const takePhotoOptions = useMemo<TakePhotoOptions & TakeSnapshotOptions>(
[... 274 lines cut ...]
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
"}, "code": "capture/inactive-source", "message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "userInfo": null}
 LOG  stopped recording video!
 LOG  Re-rendering camera page with active camera. Device: "front (1)" (3648x1640 @ 30fps)
 LOG  Camera initialized!
 LOG  Re-rendering camera page with active camera. Device: "front (1)" (3648x1640 @ 30fps)
 LOG  Re-rendering camera page with inactive camera. Device: "front (1)" (3648x1640 @ 30fps)
```

### Device

Samsung Galaxy S20

### VisionCamera Version

2.15.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vietnt-uet, 2022-11-04:
Should you try to pause the recording before switch camera and resume recording after that?

ridvanaltun, 2023-01-27:
Same issue here: https://github.com/mrousavy/react-native-vision-camera/issues/834

shubhamchepe, 2023-07-23:
Hey!,
Did you solve the issue?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

balloman, 2023-08-19:
I am still unable to get recordings to work while switching the camera. I'm gonna try to spend a couple minutes looking at it, but the app doesn't crash, the recording just ends at the point where you switch cameras

balloman, 2023-08-19:
Hmm, actually the entire recording is kept, even the audio, the video just freezes at the point of the camera switch

balloman, 2023-08-21:
Yep! With the branch you recommended.

closed 2023-09-01 by mrousavy