margelo/react-native-vision-camera #1298: Is it possible to edit/create ImageProxy image in Frame Processor Plugin and return it❓

Easy47, 2022-10-22:
### Question

I would like to be able to test the accuracy of frame processor plugins such as OCR or Barcode so I can make some image processing on the frames. (The answers to these questions will help me later to add image processing to the frames)

I would therefore like to be able to replace the frame with other images (those of my test suite) and then return it so i can give the edited frame to OCR/Barcode (or any other) frame processor. 

Is it possible to create a new ImageProxy in frame processor plugin or simply edit the image passed in parameter of the callback function ? 
If yes, how can I feed the ImageProxy object with my jpg loaded image so i can return it and use it ?

Thanks in advance.

### What I tried

To do this, I am currently trying to load an image and replace the content of the frame with the loaded image and return it so i can use the "new frame" in an other fp plugin.

```java
package com.mrousavy.camera.example;

import android.graphics.Bitmap;
import android.graphics.BitmapFactory;
import android.graphics.ImageFormat;
import android.graphics.Rect;
import android.graphics.YuvImage;
import android.util.Log;
import androidx.camera.core.ImageProxy.PlaneProxy;
import androidx.camera.core.ImageProxy;
import com.facebook.react.bridge.WritableNativeArray;
import com.facebook.react.bridge.WritableNativeMap;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
import java.io.ByteArrayOutputStream;
import java.io.File;
import java.io.FileOutputStream;
import java.nio.ByteBuffer;
import java.util.Arrays;
import org.jetbrains.annotations.NotNull;

public class ExampleFrameProcessorPlugin extends FrameProcessorPlugin {
    private void encodeYUV420SP(Bitmap image, byte[] Y, byte[] U, byte[] V) {
        int inputWidth = image.getWidth();
        int inputHeight = image.getHeight();
        int [] argb = new int[inputWidth * inputHeight];
        image.getPixels(argb, 0, inputWidth, 0, 0, inputWidth, inputHeight);

        byte [] yuv420sp = new byte[inputWidth*inputHeight*3/2];
        final int frameSize = inputWidth * inputHeight;

        int yIndex = 0;

        int a, R, G, B;
        int index = 0;
        int uvIndex = 0;
        for (int j = 0; j < inputHeight; j++) {
            for (int i = 0; i < inputWidth; i++) {

                a = (argb[index] & 0xff000000) >> 24; // a is not used obviously
                R = (argb[index] & 0xff0000) >> 16;
                G = (argb[index] & 0xff00) >> 8;
                B = (argb[index] & 0xff) >> 0;

                int Ypixel = ( (  66 * R + 129 * G +  25 * B + 128) >> 8) +  16;
                int Upixel = ( ( -38 * R -  74 * G + 112 * B + 128) >> 8) + 128;
                int Vpixel = ( ( 112 * R -  94 * G -  18 * B + 128) >> 8) + 128;

                Y[index] = (byte) ((Ypixel < 0) ? 0 : ((Ypixel > 255) ? 255 : Ypixel));
                if (j % 2 == 0 && index % 2 == 0) { 
                    V[uvIndex] = (byte)((Vpixel<0) ? 0 : ((Vpixel > 255) ? 255 : Vpixel));
                    U[uvIndex] = (byte)((Upixel<0) ? 0 : ((Upixel > 255) ? 255 : Upixel));
                    uvIndex++;
                }

                index ++;
            }
        }
        System.out.println("UVINDEX: " + uvIndex);
    }

    @Override
    public Object callback(@NotNull ImageProxy image, @NotNull Object[] params) {
        try {
            // Get path of image to load
            String param = params[0].toString();

            // Get image
            Bitmap bmImg = BitmapFactory.decodeFile(param);

            // Try to transform bitmap to Y U V array (representing their eponym channel)
            byte[] Y = new byte[bmImg.getWidth() * bmImg.getHeight()];
            byte[] U = new byte[bmImg.getWidth() * bmImg.getHeight()];
            byte[] V = new byte[bmImg.getWidth() * bmImg.getHeight()];
            encodeYUV420SP(bmImg, Y, U, V);
            PlaneProxy[] planes = image.getPlanes();

            ByteBuffer bufferY = ByteBuffer.wrap(Y);
            planes[0].getBuffer().put(bufferY);

            ByteBuffer bufferU = ByteBuffer.wrap(U);
            planes[1].getBuffer().put(bufferU);

            ByteBuffer bufferV = ByteBuffer.wrap(V);
            planes[2].getBuffer().put(bufferV);

            return image;
            //return test;
        } catch (Exception e) {
            System.out.println("CRASHED");
            e.printStackTrace();
        }
        return null;
    }

    ExampleFrameProcessorPlugin() {
        super("example_plugin");
    }
} 
```
But it s not working.

### VisionCamera Version

2.14

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-10-25:
That should work!

If you do

```ts
const fp = useFrameProcessor((frame) => {
  'worklet'

  const newFrame = example_plugin(frame)
  const whatever = scanQrCodesOrWhatever(newFrame) // <-- uses the one from jpg
}, [])

```

It should work. Doesn't it?

Easy47, 2022-10-25:
> 

Yes ! That's what I want to do.
However, I don't know how to create a new ImageProxy object from the jpeg image. (Nor how to replace the planes of the ImageProxy object which is passed as a parameter of the "callback" function)
:(

mrousavy, 2022-10-25:
Oh that's what you mean! Okay got it now. I am not 100% sure how to do that tbh, but I think we could create a constructor for `FrameHostObject` that also allows the `Image` type, then you can just return a `Image` (or `Bitmap`) and it would work, no need for `ImageProxy`.

cc @thomas-coldwell for our v3 planning

Easy47, 2022-10-25:
> Oh that's what you mean! Okay got it now. I am not 100% sure how to do that tbh, but I think we could create a constructor for `FrameHostObject` that also allows the `Image` type, then you can just return a `Image` (or `Bitmap`) and it would work, no need for `ImageProxy`.

Thank you. 
So for the moment it is not possible to modify a frame itself / to create a new one in a frame processor plugin and return it?

xulihang, 2022-11-02:
I need to return Bitmap for Android and UIImage for iOS as well, since I need to rotate and crop the frame and even run perspective transformation.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy