margelo/react-native-vision-camera #1305: Frame processors are not enabled when open the camera 🐛 🐛 🐛

KonstantinZhukovskij, 2022-10-25:
### What were you trying to do?

I get the following error when I open the camera, on both platforms. 

[image]

**It is mandatory for me to use the `use_frameworks`, please do not advise to remove it.**

```
"react": "17.0.2",
"react-native": "0.67.4",
"react-native-vision-camera": "2.15.1",
"vision-camera-code-scanner": "0.2.0",
"react-native-reanimated": "2.11.0"
```

I saw the same build error earlier in the issues here, but I managed to fix it. 
Podfile
```
require File.join(File.dirname(`node --print "require.resolve('expo/package.json')"`), "scripts/autolinking")
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, '13.0'
install! 'cocoapods', :deterministic_uuids => false

target 'charge' do
  use_expo_modules!

  post_integrate do |installer|
    begin
      expo_patch_react_imports!(installer)
      rescue => e
      Pod::UI.warn e
    end
  end

  use_frameworks! :linkage => :static
  config = use_native_modules!

  pre_install do |installer|
    Pod::Installer::Xcode::TargetValidator.send(:define_method, :verify_no_static_framework_transitive_dependencies) {}

    installer.pod_targets.each do |pod|
      if pod.name.eql?('RNPermissions') || pod.name.start_with?('Permission-') || pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera')
        def pod.build_type;
        Pod::BuildType.static_library
      end
    end
  end
end

use_react_native!(
                  :path => config[:reactNativePath],
                  :hermes_enabled => false,
                  :fabric_enabled => false,
                  :app_path => "#{Pod::Config.instance.installation_root}/.."
                  )

post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['ENABLE_BITCODE'] = 'NO'
    end

    if target.name == "vision-camera-code-scanner"
      target.build_configurations.each do |config|
        config.build_settings['ENABLE_BITCODE'] = 'NO'
        config.build_settings['ONLY_ACTIVE_ARCH'] = 'NO'
        config.build_settings['BUILD_LIBRARY_FOR_DISTRIBUTION'] = 'YES'
      end
    end
  end

  react_native_post_install(installer)
  __apply_Xcode_12_5_M1_post_install_workaround(installer)
end
end

```

### What happened instead?

The app crashes with `Frame processors are not enabled` error.

### Device

iPhone, Android

### VisionCamera Version

2.15.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

KonstantinZhukovskij, 2022-10-25:
I think this issue is related to the same error. [LINK](https://github.com/mrousavy/react-native-vision-camera/issues/1294)

KonstantinZhukovskij, 2022-10-25:
@mrousavy I need your help here)

elfinamld, 2022-10-25:
Hi, maybe you can try with this comment :) https://github.com/rodgomesc/vision-camera-code-scanner/issues/29#issuecomment-1060776135

KonstantinZhukovskij, 2022-10-25:
No, of course, this also happens on the prod version for real devices.

KonstantinZhukovskij, 2022-10-26:
I am also sure that this will be reproducible on all future versions of react. Here the problem lies precisely in the use of the **`use_frameworks`**

jtweaver, 2022-11-05:
I'm seeing the same issue. Except I don't have use_frameworks in my Podfile. React-native version 0.70.4. This should be reproducible by just using `npx react-native init` to create a new RN project and adding react-native-vision-camera and a frame processor.

jtweaver, 2022-11-05:
> I'm seeing the same issue. Except I don't have use_frameworks in my Podfile. React-native version 0.70.4. This should be reproducible by just using `npx react-native init` to create a new RN project and adding react-native-vision-camera and a frame processor.

Scratch that. I got it fixed by running `npx react-native-clean-project`

KonstantinZhukovskij, 2022-11-08:
Obviously, the question here is in the **use_framework** :)

valn1, 2022-11-08:
> Obviously, the question here is in the **use_framework** :)

do you really need  use_frameworks ?
i say this becaus i also thought i needed it for firebase and stuff untill i got tired and downgraded it to 15.3.0 and did this to my podfile:

    pod 'Firebase', :modular_headers => true
    pod 'FirebaseCore', :modular_headers => true
    pod 'FirebaseCoreInternal', :modular_headers => true
    pod 'GoogleUtilities', :modular_headers => true

KonstantinZhukovskij, 2022-11-08:
You can use the **use_framework** not only for Firebase and downgrading is not a solution

Simoon-F, 2022-11-21:
> > 显然，这里的问题是在**use_framework** :)
> 
> 你真的需要 use_frameworks 吗？ 我这么说是因为我还认为我需要它用于 firebase 和其他东西，直到我厌倦并将它降级到 15.3.0 并对我的 podfile 执行此操作：
> 
> ```
> pod 'Firebase', :modular_headers => true
> pod 'FirebaseCore', :modular_headers => true
> pod 'FirebaseCoreInternal', :modular_headers => true
> pod 'GoogleUtilities', :modular_headers => true 
> ```

You don't need to downgrade, you can also use `firebase` in `use_frameworks! :linkage => :static`

Simoon-F, 2022-11-21:
> 显然，这里的问题是在**use_framework** :)

Have you solved it?

VinhKyiT, 2023-03-21:
> > 显然，这里的问题是在**use_framework** :)
> 
> Have you solved it?

I can use this syntax with Firebase Storage, It might required to use the use_framework!

valn1, 2023-04-10:
https://github.com/mrousavy/react-native-vision-camera/issues/1281#issuecomment-1433243369
@KonstantinZhukovskij  try this, i did some witchcraft and found a solution, its not short by any means, but me and some others found it useful.
hope it helps

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy