margelo/react-native-vision-camera #1312: 🐛 'boost/operators.hpp' file not found

xulihang, 2022-10-31:
### What were you trying to do?

Build for Android

### What happened instead?

It produces the following error message:

```
C:\Users\admin\AppData\Local\Android\Sdk\ndk\21.4.7075529\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=armv7-none-linux-androideabi21 --gcc-toolchain=C:/Users/admin/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/admin/AppData/Local/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/windows-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.3.0-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/React -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/React/Base -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactAndroid/src/main/jni -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactCommon -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactCommon/callinvoker -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactCommon/jsi -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/hermes-engine/android/include -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/AnimatedSensor -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/Tools -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/Registries -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/AnimatedSensor -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/Tools -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/SpecTools -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/SharedItems -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/Registries -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/LayoutAnimations -IF:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -DJS_RUNTIME_HERMES=1 -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o -MF CMakeFiles\VisionCamera.dir\src\main\cpp\JSIJNIConversion.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o -c ../../../../src/main/cpp/JSIJNIConversion.cpp
  In file included from ../../../../src/main/cpp/JSIJNIConversion.cpp:16:
  In file included from F:/test/react-native/vision-camera-dynamsoft-document-normalizer/example/node_modules/react-native/ReactAndroid/src/main/jni\react/jni/NativeMap.h:11:
  ../../../../build/third-party-ndk/folly\folly/dynamic.h:63:10: fatal error: 'boost/operators.hpp' file not found
  #include <boost/operators.hpp>
```

### Device

Oppo

### VisionCamera Version

2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jad-Jbara, 2022-11-01:
Upgrading to 2.15.2 resolves this issue

xulihang, 2022-11-02:
Resovled by updaing to 2.15.2.

closed 2022-11-02 by xulihang