margelo/react-native-vision-camera #1314: 🐛  Blank white screen on ios device

ZerakPalani, 2022-11-01:
### What were you trying to do?

I take a picture and I present PreviewPage(MediaPage.tsx) and boom, it's just white blank screen on my physical device (iphone 12 pro, ios 16), on ios simulator I tried to get a error by presenting this page with a file path & type as parameters and it worked on the ios simulator. But after looking at the code I think the issue somewhat has to do with mediahasloaded process. Like it can't load it even though I take the photo? I even made an alert to log the path and type and I get path: /private/...jpg type "photo" which is pretty valid. But something seems to not work on the physical device?
For the camera component, I had to remove frame processors otherwise it was just a white blank page there too. But in MediaPage there doesn't seem to be anything related to frame processors, etc. So any idea of what the issue could be?

Thank you.

### What happened instead?

Just a white blank page.

### Device

iPhone 12 pro

### VisionCamera Version

^2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Raghavmunjal, 2022-11-16:
getting same issue on v2.13.0 as well any Solution to this?
@mrousavy

ZerakPalani, 2022-11-16:
> getting same issue on v2.13.0 as well any Solution to this? @mrousavy

Sadly nothing from what I have checked... I even removed the package and functionality from my app and kept coming back to the issues list to see if there's any solution or what the cause could be.. Strangely when I tried to debug it, since camera doesn't work on ios simulator, I downloaded it on physical device and made an alert to show me the output sent to media, I followed same object structure with path and type on the ios simulator and skipped the camera part, instead I used a path of a image on the ios simulator (from media gallery). And then it worked (didn't get the white blank screen), but not when I take a photo or video from physical device, it just becomes blank.

ChauVV, 2022-12-29:
event on android get this error too

derb21, 2023-02-02:
Probably need to add file:// before like this 'file://' + photo.path

ZerakPalani, 2023-02-03:
> Probably need to add file:// before like this 'file://' + photo.path
Already tried that...

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy