margelo/react-native-vision-camera #1316: 🐛 Frame Processor threw an error: Cannot read property '__callAsync' of undefined

xulihang, 2022-11-02:
### What were you trying to do?

Use runOnJS in a frameProcessor.

I am using "react-native-reanimated": "^2.9.1".

### What happened instead?

It produces the following error:

```
Frame Processor threw an error: Cannot read property '__callAsync' of undefined
```

### Device

OPPO A72

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2022-11-02:
I am having this issue with hermes disabled. 

```
A/libc: Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x730231cc08 in tid 20761 (pool-6-thread-1), pid 19866 (umentnormalizer)
```

xulihang, 2022-11-02:
Since I cannot use `runOnJS`, I have to pass values from the plugin to JS using useSharedValue.

xulihang, 2022-11-02:
`runOnJS` actually works fine in useDerivedValue.

```ts
const pointsData = REA.useDerivedValue(() => {
    let data = "";
    REA.runOnJS(setPointsText)(data);
    console.log(data);
    return data;
  }, [detectionResults]);
```

rafaelmotta, 2023-06-07:
I was stuck in this problem for a while. I resolved by declaring the function called in `runOnJS` before the `useFrameProcessor` declaration

Before (with error):

```typescript
const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';

        const lorem = true;
        runOnJS(handleLorem)(true);
    },
    [detector],
  );

const handleLorem = (lorem: boolean) => { ... }
```

Later (works):

```typescript
const handleLorem = (lorem: boolean) => { ... }

const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';

        const lorem = true;
        runOnJS(handleLorem)(true);
    },
    [detector],
  );
```

rafaelmotta, 2023-06-07:
I think we could close this one

closed 2023-06-07 by xulihang

fukemy, 2023-07-27:
hi @xulihang , did u solved this?