margelo/react-native-vision-camera #1317: 🐛 zoom reset automatically when camera component re-renders

jasbirrajrana, 2022-11-03:
### What were you trying to do?

I want to set zoom=2 without a pinch gesture.

### Reproduceable Code

```tsx
const QRScanner=()=>{
const devices = useCameraDevices();
const device = devices.back;
const onQRCodeDetected = useCallback((qrCodeText: string) => {
        onQrRead(qrCodeText, true, { setIsCameraActive }); 
       // I have an onQrRead Function that takes a string input and calls APIs accordingly.
    }, []);

    useFocusEffect(
        useCallback(() => {
            setIsCameraActive(true);
        }, []),
    );

    const frameProcessor = useFrameProcessor(frame => {
        'worklet';

        const detectedQrCodes = scanBarcodes(frame, types, options);
        if (detectedQrCodes?.length > 0 && detectedQrCodes?.[0]?.displayValue && isCameraActive) {
            runOnJS(setIsCameraActive)(false);
            runOnJS(onQRCodeDetected)(detectedQrCodes?.[0]?.displayValue);
        }
    }, []);
return (
<Camera
      style={styles.cameraView}
      device={device}
      isActive
      video={false}
      torch={torch}
      zoom={2}
      frameProcessor={frameProcessor}
/>
)
}
```

### What happened instead?

zoom Automatically got resets when the camera component gets re-renders.

### Device

samsung A23 (Andriod)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jasbirrajrana, 2022-11-03:
submitting issue first time let me know if something require!

prabucki, 2023-07-27:
Still seeing this issue in 2.15.4.

Seems to only happen when `isActive` is changed.

One workaround is to change the zoom value after re-render to anything other than what it was previously set to. So, looking at the code from the previous comment, re-setting zoom value as `2` wouldn't work, but anything other like `1.99999` or `2.00001` triggers the zoom in the camera to be set correctly instead of defaulting to `1`. 

Looking at the vision camera code, it seems that `shouldReconfigureZoom` in `react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt` should hit `true` when the re-render happens but it doesn't because the zoom prop stays the same:
```kotlin
val shouldReconfigureZoom = shouldReconfigureSession || changedProps.contains("zoom")
``` 
Changing this to:
```kotlin
val shouldReconfigureZoom = true
``` 

fixed the issue for me, but I've got no clue how it affects the rest of the application - looks hacky.

Hope this helps.

alessandrom, 2023-07-31:
Also in my case the zoom is resetting automatically after I take a picture (I had changed zoom with the gesture).
The issue seems bigger on Android where it's impossible to take a zoomed picture on my device

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy