margelo/react-native-vision-camera #1319: ✨ Make startRecording() async

elirichey, 2022-11-05:
### What feature or enhancement are you suggesting?

Hey Marc, this package is amazing and I appreciate all the time and effort you've put into it! 

I noticed this in your ToDo file:

* [ ] Make `startRecording()` async. Due to NativeModules limitations, we can only have either one callback or one promise in a native function. For `startRecording()` we need both, since you probably also want to catch any errors that occurred during a `startRecording()` call (or wait until the recording has actually started, since this can also take some time)

Any thoughts on when this might be completed? I'm trying to get an accurate timestamp from when video recording actually starts. Without it being async, the timestamps are off by about 1.3-1.6 seconds in iOS. I haven't tested for difference in Android yet. Will do shortly and update findings then. 

Thanks again,

Eli

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I'm currently testing on a couple of devices to see if I can find a standard offset. Considering I'm still fact-finding I don't have any exact answers. Though, I imagine this approach will be fruitless. Hard-coding offsets doesn't seam like a viable solution, but I'm going to try it anyway. 

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-09-30:
I think such an API makes more sense:

```ts
const recorder = await camera.current.prepareRecorder()

await recorder.start()
await recorder.pause()
console.log(recorder.duration)
await recorder.resume()
await recorder.stop()
```

..exposed as a HostObject using JSI. That way you can also prepare the recorder beforehand to warm it up.

mrousavy, 2023-11-21:
Started working on something here: https://github.com/mrousavy/react-native-vision-camera/pull/2194

as of now this doesn't really work, but we'll see what we can do