margelo/react-native-vision-camera #1322: 🐛 AndroidRuntime when trying to run Android app - No virtual method runOnJSQueueThread

dittmarconsulting, 2022-11-06:
### What were you trying to do?

I have successfully installed the package on iOS with the following setup

```
"react": "17.0.2",
"react-native": "0.67.0",
"react-native-reanimated": "^2.11.0",
"react-native-vision-camera": "^2.14.1",
```

but I have trouble getting to run the app on Android. The app crashes at start with the following logcat output

```
2022-11-06 19:16:25.982 9363-9425/au.com.kolmeo.pro E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-1
    Process: au.com.testapp, PID: 9363
    java.lang.NoSuchMethodError: No virtual method runOnJSQueueThread(Ljava/lang/Runnable;)Z in class Lcom/facebook/react/bridge/ReactApplicationContext; or its super classes (declaration of 'com.facebook.react.bridge.ReactApplicationContext' appears in /data/app/au.com.testapp-IAj_rW6-mQJpuMop7L3yzg==/base.apk)
        at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.<init>(FrameProcessorRuntimeManager.kt:53)
        at com.mrousavy.camera.CameraViewModule.initialize$lambda-0(CameraViewModule.kt:66)
        at com.mrousavy.camera.CameraViewModule.lambda$YiRnUrObfbO8oH48lnR0UOsrros(Unknown Source:0)
        at com.mrousavy.camera.-$$Lambda$CameraViewModule$YiRnUrObfbO8oH48lnR0UOsrros.run(Unknown Source:2)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
```

It seems the Kotlin code of the `FrameProcessorRuntime` wants to invoke this function

```
context.runOnJSQueueThread {
  installJSIBindings()
}
```

but it can't find `runOnJSQueueThread`.

I'm normally pretty good in debugging this kind of stuff but when it comes to JSI I'm a bit of a noob TBH.

### Reproduceable Code

```tsx
Install as per install-guide of `react-native-reanimated` & `react-native-vision-camera`
```

### What happened instead?

The app crashes at startup.

### Relevant log output

```shell
2022-11-06 19:16:25.982 9363-9425/au.com.kolmeo.pro E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-1
    Process: au.com.testapp, PID: 9363
    java.lang.NoSuchMethodError: No virtual method runOnJSQueueThread(Ljava/lang/Runnable;)Z in class Lcom/facebook/react/bridge/ReactApplicationContext; or its super classes (declaration of 'com.facebook.react.bridge.ReactApplicationContext' appears in /data/app/au.com.testapp-IAj_rW6-mQJpuMop7L3yzg==/base.apk)
        at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.<init>(FrameProcessorRuntimeManager.kt:53)
        at com.mrousavy.camera.CameraViewModule.initialize$lambda-0(CameraViewModule.kt:66)
        at com.mrousavy.camera.CameraViewModule.lambda$YiRnUrObfbO8oH48lnR0UOsrros(Unknown Source:0)
        at com.mrousavy.camera.-$$Lambda$CameraViewModule$YiRnUrObfbO8oH48lnR0UOsrros.run(Unknown Source:2)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
```

### Device

Emulator Pixel 2, Android 10

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

UrzaDestiny, 2022-11-06:
Facing the same issue(

Joshig99, 2022-11-06:
Hi, I'm also getting same error

Joshig99, 2022-11-06:
Hi, seems its a react-native-reanimated package issue, this solution worked for me [](https://github.com/software-mansion/react-native-reanimated/issues/3743#issuecomment-1304437673)

dittmarconsulting, 2022-11-07:
@Joshig99 

I can confirm that the solution you linked to worked for me too. 

Perhaps the react-native-vision-camera team can have a look at this or add the solution to the install documents.

Cheers @Joshig99

zumaawad, 2022-11-07:
I'm also getting same error, but this fix works for me
https://github.com/mrousavy/react-native-vision-camera/issues/1322#issuecomment-1304812239

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy