margelo/react-native-vision-camera #1323: is this  package support new react native Architecture ❓

hari-mohan-choudhary, 2022-11-06:
### Question

I have enable react native new react native Architecture. I am not able to use this package. Camera is null.

### What I tried

I am using react native 0.70.4. 

### VisionCamera Version

0.70.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

hari-mohan-choudhary, 2022-11-08:
anyone can help me on this

BraveEvidence, 2023-02-24:
I have managed to get scan QR code / barcode with new architecture. Check this https://github.com/BraveEvidence/ScanQrFabricReactNative-sNewArchitecture2023/tree/main , https://www.youtube.com/watch?v=q4IiUIARANM&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=9 and https://www.youtube.com/watch?v=sE0MEyFA7E4&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=10

DominikHinc, 2023-08-09:
Hey,
I have prepared an initial support for Fabric here: https://github.com/mrousavy/react-native-vision-camera/pull/1684

mrousavy, 2023-09-30:
Hey, not yet.

closed 2023-09-30 by mrousavy