margelo/react-native-vision-camera #1324: ❓ No objects detected on iOS with Frame Processor

GaylordP, 2022-11-07:
### Question

Hello and thank you for your fabulous library :)

Sorry for my English, I'm French. :)

I'm trying to create a [Frame Processor](https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors) (which I may publish [in the plugins](https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processor-plugin-list)) that can detect an object, frame it with a border and take a picture of that frame.

On Android, everything works fine : I manage to detect all the objects [with a .java plugin](https://developers.google.com/ml-kit/vision/object-detection/android), then frame and trim them in .jsx :)

But on iOS, no object is detected, ever. I've been trying for more than four days but without success.

### What I tried

```
#import <Foundation/Foundation.h>
#import <VisionCamera/FrameProcessorPlugin.h>
#import <VisionCamera/Frame.h>
#import <MLKit.h>

@interface VisionScanObjectsFrameProcessorPlugin : NSObject
+ (MLKObjectDetector*) objectDetector;
@end

@implementation VisionScanObjectsFrameProcessorPlugin

+ (MLKObjectDetector*) objectDetector {
  static MLKObjectDetector* objectDetector = nil;
  if (objectDetector == nil) {
    MLKObjectDetectorOptions *options = [[MLKObjectDetectorOptions alloc] init];
    options.detectorMode = MLKObjectDetectorModeSingleImage;
    options.shouldEnableMultipleObjects = YES;
    options.shouldEnableClassification = YES;
    
    objectDetector = [MLKObjectDetector objectDetectorWithOptions:options];
  }

  return objectDetector;
}

static inline id scanObjects(Frame* frame, NSArray* arguments) {
  MLKVisionImage *image = [[MLKVisionImage alloc] initWithBuffer:frame.buffer];
  image.orientation = frame.orientation; // <-- TODO: is mirrored?

  NSError* error;
  NSArray<MLKObject*>* objects = [[VisionScanObjectsFrameProcessorPlugin objectDetector] resultsInImage:image error:&error]; // I also tried with processImage: https://developers.google.com/ml-kit/vision/object-detection/ios#3.-process-the-image
  
  NSLog(@"Object detected : %ld", objects.count);
  
  NSMutableArray* results = [NSMutableArray arrayWithCapacity:objects.count];
  for (MLKObject* object in objects) {
    NSMutableArray* labels = [NSMutableArray arrayWithCapacity:object.labels.count];

    for (MLKObjectLabel* label in object.labels) {
        [labels addObject:@{
          @"index": [NSNumber numberWithFloat:label.index],
          @"label": label.text,
          @"confidence": [NSNumber numberWithFloat:label.confidence]
        }];
    }
  }

  return results;
}

VISION_EXPORT_FRAME_PROCESSOR(scanObjects)

@end
```

When I test this simple code, I always have in my console `Object detected : 0`. Exceptionally, I will have an object detected but it is very rare.

On Android, I have hundreds of objects detected; the code is identical (but adapted to java).

Do you have any clues as I really feel I'm reaching the end of my attempts :/
Thanks in advance :)

### VisionCamera Version

^2.15.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

hassanrehman01398, 2022-11-26:
HI. Can you share your repo so that I can look at the problem in IOS?

GaylordP, 2022-11-26:
Hi

Thank you for your answer :)
I did it by reducing the size of the frames, it's ok now :)

closed 2022-11-26 by GaylordP

hassanrehman01398, 2022-11-26:
Can you share your repo please as I am stuck with the integration of custom model with it.

GaylordP, 2022-11-26:
Google doc : https://developers.google.com/ml-kit/vision/image-labeling/custom-models/ios
Here is the my iOS/Swift detection code, limited to certain objects: 

```
//
//  VisionScanObjectsFrameProcessorPlugin
//
//  Par gaylord pour détection des factures à la caméra
//

import Foundation
import AVFoundation
import MLKit

@objc(VisionScanObjectsFrameProcessorPlugin)
public class VisionScanObjectsFrameProcessorPlugin: NSObject, FrameProcessorPluginBase {
  static var ObjectDetectorOptions: CustomObjectDetectorOptions = {
    let localModelFilePath = Bundle.main.path(forResource: "lite-model_object_detection_mobile_object_labeler_v1_1", ofType: "tflite")
    let localModel = LocalModel(path: localModelFilePath!)

    let options = CustomObjectDetectorOptions(localModel: localModel)
    options.detectorMode = .singleImage
    options.shouldEnableClassification = true
    options.shouldEnableMultipleObjects = true
    options.classificationConfidenceThreshold = NSNumber(value: 0.5)
    options.maxPerObjectLabelCount = 3
    return options
  }()
 
  static var objectDetector  = ObjectDetector.objectDetector(options: ObjectDetectorOptions)

  @objc
  public static func callback(_ frame: Frame!, withArgs _: [Any]!) -> Any! {
    let frameResizedToUiimage = resizeFrameToUiimage(frame: frame)!

    let image = VisionImage(image: frameResizedToUiimage)
    image.orientation = imageOrientation(
      deviceOrientation: UIDevice.current.orientation,
      cameraPosition: AVCaptureDevice.Position.back
    )

    let resultsArray: NSMutableArray = []

    var objects: [Object]

    do {
        objects = try objectDetector.results(in: image)
    } catch _ {
        return []
    }

    for object in objects {
      let labelsArray: NSMutableArray = []
      
      for label in object.labels {
        if (122 == label.index || 188 == label.index || 288 == label.index || 325 == label.index || 357 == label.index || 370 == label.index || 480 == label.index || 510 == label.index || 551 == label.index) {
          let labelObject: NSMutableDictionary = [:]
          labelObject["index"] = label.index
          labelObject["text"] = label.text
          labelObject["confidence"] = label.confidence
          labelsArray.add(labelObject)
        }
      }

      if (0 != labelsArray.count) {
        let resultObject: NSMutableDictionary = [:]
        resultObject["width"] = object.frame.size.width
        resultObject["height"] = object.frame.size.height
        resultObject["top"] = object.frame.origin.y
        resultObject["left"] = object.frame.origin.x
        resultObject["frameResizedWidth"] = frameResizedToUiimage.size.width
        resultObject["frameResizedHeight"] = frameResizedToUiimage.size.height
        resultObject["labels"] = labelsArray
        resultsArray.add(resultObject)
      }
    }

    return resultsArray
  }

  public static func resizeFrameToUiimage(frame: Frame) -> UIImage! {
    let targetSize = CGSize(width: 224.0, height: 224.0)

    let imageBuffer = CMSampleBufferGetImageBuffer(frame.buffer)!
    let ciimage = CIImage(cvPixelBuffer: imageBuffer)
    
    let context = CIContext(options: nil)
    let cgImage = context.createCGImage(ciimage, from: ciimage.extent)!
    let uiimage = UIImage(cgImage: cgImage)

    let widthRatio  = targetSize.width  / uiimage.size.width
    let heightRatio = targetSize.height / uiimage.size.height
        
    var newSize: CGSize
    if(widthRatio > heightRatio) {
      newSize = CGSize(width: uiimage.size.width * heightRatio, height: uiimage.size.height * heightRatio)
    } else {
      newSize = CGSize(width: uiimage.size.width * widthRatio, height: uiimage.size.height * widthRatio)
    }
        
    let rect = CGRect(origin: .zero, size: newSize)
        
    UIGraphicsBeginImageContextWithOptions(newSize, false, 1.0)
    uiimage.draw(in: rect)

    let newImage = UIGraphicsGetImageFromCurrentImageContext()
      UIGraphicsEndImageContext()

    return newImage
  }

  public static func imageOrientation(
    deviceOrientation: UIDeviceOrientation,
    cameraPosition: AVCaptureDevice.Position
  ) -> UIImage.Orientation {
    switch deviceOrientation {
      case .portrait:
        return cameraPosition == .front ? .leftMirrored : .right
      case .landscapeLeft:
        return cameraPosition == .front ? .downMirrored : .up
      case .portraitUpsideDown:
        return cameraPosition == .front ? .rightMirrored : .left
      case .landscapeRight:
        return cameraPosition == .front ? .upMirrored : .down
      case .faceDown, .faceUp, .unknown:
        return .up
    }
  }
}
```

hassanrehman01398, 2022-11-26:
ok and what about android code that you used?

GaylordP, 2022-11-26:
This is for Android : 

However, resizing to Bitmap (resizeFrameToBitmap function) is a problem on some phones. I have disabled it for the moment and the result is satisfactory (= the detection works even without frame resizing)

```
package com.XXXXXX_mobile_v3.visionscanobjectsframeprocessor;

import android.annotation.SuppressLint;
import android.graphics.Bitmap;
import android.graphics.BitmapFactory;
import android.graphics.ImageFormat;
import android.graphics.Rect;
import android.graphics.YuvImage;
import android.media.Image;

import androidx.camera.core.ImageProxy;

import com.facebook.react.bridge.WritableNativeArray;
import com.facebook.react.bridge.WritableNativeMap;
import com.google.android.gms.tasks.Task;
import com.google.android.gms.tasks.Tasks;
import com.google.mlkit.common.model.LocalModel;
import com.google.mlkit.vision.common.InputImage;
import com.google.mlkit.vision.objects.custom.CustomObjectDetectorOptions;
import com.google.mlkit.vision.objects.DetectedObject;
import com.google.mlkit.vision.objects.DetectedObject.Label;
import com.google.mlkit.vision.objects.ObjectDetection;
import com.google.mlkit.vision.objects.ObjectDetector;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import org.jetbrains.annotations.NotNull;

import java.io.ByteArrayOutputStream;
import java.lang.Exception;
import java.nio.ByteBuffer;
import java.util.List;

public class VisionScanObjectsFrameProcessorPlugin extends FrameProcessorPlugin {
  private final ObjectDetector objectDetector = getObjectDetector();

  @Override
  public Object callback(ImageProxy frame, @NotNull Object[] params) {
    @SuppressLint("UnsafeOptInUsageError")

    Image mediaImage = frame.getImage();
    if (mediaImage != null) {
      InputImage image =
              InputImage.fromMediaImage(mediaImage, frame.getImageInfo().getRotationDegrees());

      Task<List<DetectedObject>> task = objectDetector.process(image);
      try {
        List<DetectedObject> detectedObjects = Tasks.await(task);
        WritableNativeArray array = new WritableNativeArray();
        for (DetectedObject detectedObject : detectedObjects) {
          WritableNativeArray labelsArray = new WritableNativeArray();

          for (Label label : detectedObject.getLabels()) {
            int index = label.getIndex();

            WritableNativeMap labelMap = new WritableNativeMap();
            if (122 == index || 188 == index || 288 == index || 325 == index || 357 == index || 370 == index || 480 == index || 510 == index || 551 == index) {
              labelMap.putInt("index", index);
              labelMap.putString("label", label.getText());
              labelMap.putDouble("confidence", label.getConfidence());

              labelsArray.pushMap(labelMap);
            }
          }

          if (0 != labelsArray.size()) {
[... 73 lines cut ...]

  private ObjectDetector getObjectDetector() {
    LocalModel localModel =
      new LocalModel.Builder()
        .setAssetFilePath("lite-model_object_detection_mobile_object_labeler_v1_1.tflite")
        .build();

    CustomObjectDetectorOptions customObjectDetectorOptions =
      new CustomObjectDetectorOptions.Builder(localModel)
        .setDetectorMode(CustomObjectDetectorOptions.STREAM_MODE)
        .enableClassification()
        .setClassificationConfidenceThreshold(0.5f)
        .setMaxPerObjectLabelCount(3)
        .build();

    ObjectDetector createObjectDetector = ObjectDetection.getClient(customObjectDetectorOptions);

    return createObjectDetector;
  }

  public VisionScanObjectsFrameProcessorPlugin() {
    super("scanObjects");
  }
}
```

GaylordP, 2022-11-26:
I may create a react-native-vision-camera plugin in mid-December, but for the moment I have the time :/

hassanrehman01398, 2022-11-26:
This is great. I am trying to use my object detection tflite model with the react-native-vision-camera plugin to make real-time detection. Let's see how much I can achieve. Thanks for the help.

hassanrehman01398, 2022-11-28:
@GaylordP Sorry to disturb you but I am facing an error in the build. Can you send me the project you worked on? The drive link will work.

CodingWithNobody, 2023-04-30:
Try this https://www.youtube.com/watch?v=ybiMSEjl5mA