margelo/react-native-vision-camera #1325: why the camera is blurry❓

Milker90, 2022-11-07:
### Question

When I scan the QR code with the camera, the camera is blurry in Huawei 30S. I use another Huawei phone and vivo phone and it can be used normally. 

device:
Huawei 30S HarmonyOS 2.0.0 

libs:
react-native: 0.70.1
react-native-vision-camera: 2.14.1
vision-camera-code-scanner: 0.2.0

code:
```js
const devices = useCameraDevices();
const device = devices.back;
const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: true,
});

<Camera
style={StyleSheet.absoluteFill}
device={device}
isActive={true}
preset="high"
frameProcessor={frameProcessor}
frameProcessorFps={5}/>
```

result:

https://user-images.githubusercontent.com/3938006/200357616-b4668900-49f3-4fb0-95aa-07b0c80332d2.mp4

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-11-22:
lol that's weird. This might be fixed with migration to VisionCamera V3, we are actively looking for companies that are willing to help fund the development of that. lmk if you're interested

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

den1a, 2024-03-06:
Hi. The same problem with the version 3.9.1
If you change the size of the camera and save it, the sharpness appears

https://github.com/mrousavy/react-native-vision-camera/assets/45651923/d712fa0b-f4a1-4558-af6c-302e5fbfb551

jkester1986, 2025-02-17:
@mrousavy we are having the same issue - camera viewer quality is awful (iPhone Xs) but image is fine when captured

ralppppy, 2025-04-25:
Still having the same issue on iPhones, how can we solve this?

justb, 2025-06-13:
@mrousavy same issue with 4.x