margelo/react-native-vision-camera #1327: 🐛 Image orientation of takePhoto

xulihang, 2022-11-08:
### What were you trying to do?

I try to take photo on both Android and iOS.

### What happened instead?

On Android, the photo taken in portrait mode is rotated, while on iOS, the photo taken in portrait mode is not rotated.

### Device

iPhone SE 2016; OPPO A72

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2022-11-08:
I checked again. The image is saved in the correct orientation on iOS. It's just my image processing library uses the orientation exif info which treat it as a landscape image.

closed 2022-11-08 by xulihang

HenrikZabel, 2022-12-19:
How did you change it?

I have the same problem. I just have the Vision Camera and no other library for the process of taking an image.

xulihang, 2022-12-20:
I use the following code: https://github.com/tony-xlh/vision-camera-dynamsoft-document-normalizer/blob/a666a84ee1c8c9128ef4afe2293ba31cb207f502/ios/BitmapUtils.swift#L13

HenrikZabel, 2022-12-20:
Thank you. I had the problem on Android too. What I did: instead of using takePhoto() I now use takeSnapshot().

It's not optimal but my App isn't a photo App anyway so the quality should be totally fine