margelo/react-native-vision-camera #1328: ❓ Is it possible to render the camera partially

yittoo, 2022-11-08:
### Question

We would like to render camera just partially inside view however camera is rendering 100% device width/height behind the scenes and is overlapping over the `Status Bar` as can be seen in screenshot. Is there a way to write in such a way that we only take the displayed segment into consideration only?

[image]

### What I tried

We tried `style` and `format` props and changing these but could not quite get it to work.

Relevant Library Versions:
react-native - 0.70.3
react-native-vision-camera: 2.15.2
react - 18.1.0
react-native-reanimated - 2.12.0

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

DomGarza, 2022-11-25:
Why would you do such a thing 🥴

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

Preview now has a resizeMode btw

closed 2023-09-30 by mrousavy