margelo/react-native-vision-camera #1330: 🐛 Mute phone call sound on Android

NataliaNobre, 2022-11-08:
### What were you trying to do?

When I'm recording a video on Android and I receive a phone call the sound of the phone ringing is on the recording.

This was fixed for iOS on [this release](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.15.0) but the issue continues on Android.

### What happened instead?

Instead of continuing the recording with no sound of the phone ringing you can hear the phone notification on the video.

### Device

Samsung S20

### VisionCamera Version

2.15.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

franbe7, 2022-11-17:
I'm having the same issue

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy