margelo/react-native-vision-camera #1332: ❓ Photo resolution using takePhoto

xulihang, 2022-11-09:
### Question

I am trying the `takePhoto` method.

I can get a high-resolution photo on Android while on iOS, I can only get a photo with a resolution of 1920x1080.

Is there a way to take a photo with a higher resolution on iOS?

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
See `useCameraFormat` API in docs

closed 2023-09-30 by mrousavy