margelo/react-native-vision-camera #1333: ❓ Stream camera video to RTSP server in H.265

jim-king-2000, 2022-11-09:
### Question

We are going to develop an app to:
1. Capture built-in camera stream.
1. Encoding the stream in H.265 with hardware acceleration.
1. Push the stream to a RTSP server.

Shall we use `react-native-vision-camera)` to implement it?

### VisionCamera Version

2.0.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Sid-Turner-Ellis, 2023-01-27:
Did you end up getting this to work @jim-king-2000 ?

jim-king-2000, 2023-01-30:
> Did you end up getting this to work @jim-king-2000 ?

No.

Sid-Turner-Ellis, 2023-01-30:
Are you using frame processors for this? How are you getting access to the video stream at all?

jim-king-2000, 2023-01-30:
> Are you using frame processors for this? How are you getting access to the video stream at all?

1. What is "frame processors"?
1. We use `react-native-webrtc` to access the video stream.

Sid-Turner-Ellis, 2023-01-30:
Thanks for letting me know, I was considering using frame processors to do this - check them out in the docs if you haven't heard of them, they seem to allow you access to each frame from the video feed

jim-king-2000, 2023-01-30:
> Thanks for letting me know, I was considering using frame processors to do this - check them out in the docs if you haven't heard of them, they seem to allow you access to each frame from the video feed

I'm not the expert of video. But I was heard that frame processor is used to process video frames such as adding rectangles to human faces. For H.265 hardware encoding on Android, `MediaCodec` is widely used.

And there is still another issue for using this great library. We write a RTSP client by RN and it would be killed by Android system soon. We don't have any idea to keep the RN process alive.

jim-king-2000, 2023-01-30:
> > Thanks for letting me know, I was considering using frame processors to do this - check them out in the docs if you haven't heard of them, they seem to allow you access to each frame from the video feed
> 
> I'm not the expert of computer video. But I was heard that frame processor is used to process video frames such as adding rectangles to human faces. For H.265 hardware encoding on Android, `MediaCodec` is widely used.
> 
> And there is still another issue for using this great library. We write a RTSP client by RN and it would be killed by Android system soon (on a device, not a smartphone). We don't have any idea to keep the RN process alive.

Sid-Turner-Ellis, 2023-03-07:
The docs and a few issues do say they can be used for sending video e.g. https://github.com/mrousavy/react-native-vision-camera/issues/965.

How are you using react-native-rtc to access the stream? 

Also your second issue sounds a lot like my issue... I outlined it here: https://github.com/mrousavy/react-native-vision-camera/issues/1011#issuecomment-1439611038. Right now i'm just handling the situations where the app is closed, its not ideal but I don't think its preventable.