margelo/react-native-vision-camera #1335: 🐛  Camera turns on but stops sometimes. how to solve it?

tndus77, 2022-11-09:
### What were you trying to do?

It was the same when using react-native-vision-camera@^2.12.2.

react-native-cli: 2.0.1
react-native: 0.67.4
react-native-vision-camera@^2.15.2

https://user-images.githubusercontent.com/77282190/200793452-df976f8e-baa6-42aa-bedd-f4bcaf26c37c.MP4

### What happened instead?

The title says it all

### Device

iphone 11

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vozaldi, 2022-11-15:
It may be caused by some component constantly updating itself and caused the camera component to restart.

What I can see in your video is the camera component is unmounted briefly and mounted again.

Productivix, 2022-12-04:
hi, have you seen that topic about : I had the same and it solves my problem : 
https://mrousavy.com/react-native-vision-camera/docs/guides/lifecycle/#the-isactive-prop

CodingWithNobody, 2023-04-15:
Try this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy