margelo/react-native-vision-camera #1342: Camera.getAvailableCameraDevices() result does not match docs and too long to be useful

XaviSmith, 2022-11-16:
### What were you trying to do?

Tried using Camera.getAvailableCameraDevices() like in the example in the docs as I'm adding this to a class component and can't use the useCameraDevices() hook, but instead of "device.devices" being returned with the types of devices available to the phone, I instead got an incredibly long list of "formats" that doesn't match anything in the docs or issues section. 

Tried with both a Samsung Galaxy S8 and iPhone 8 and got similar results. Relevant bits of the code posted below

### Reproduceable Code

```tsx
import React, { Component } from 'react';
import { Camera, CameraPermissionStatus, useCameraDevices } from 'react-native-vision-camera';

class Dyno extends Component {

  
  constructor (props) {
    super(props)
    this.state = {
      viewPhoto: false,
      photoUri: null,
      modalContent: null,
      showLoading: false,
      device: null
    }
  }
  
  componentDidMount()
  { 
    this.getCameras();
    this.getCameraPermissions();
  }
  
  async getCameras()
  {
    const device = await Camera.getAvailableCameraDevices();
    console.warn("DEVICES: " + JSON.stringify(device));
    console.warn("DEVICES2 " + device.devices)
  }
  
  getCameraPermissions()
  {
        Camera.getCameraPermissionStatus().then(res => {
          console.warn('CAM RESULT: ' + res);
          if(res != 'authorized')
          {
            this.authorizeCamera();
          }
        })
  }
  
  authorizeCamera()
  {
      if(Platform.OS == 'ios')
      {
        Camera.requestCameraPermission().then(res => {
          if(res != authorized)
          {
            Alert.alert(strings.alert, "Camera not authorized! This component cannot work without camera permissions!");
          }
        })
      } else {
        Camera.requestCameraPermission().then(res => {
          if(res != authorized)
          {
            Alert.alert(strings.alert, "Camera not authorized! This component cannot work without camera permissions!");
          }
        })
      }
  }
```

### What happened instead?

Output for the "DEVICES:" log as follows on the Galaxy S8 (iphone gets a similarly formatted and long log):  https://jsonblob.com/1042496352801800192 

Output for "DEVICES2" is just "undefined" 

### Relevant log output

```shell
Using a link as the log output is incredibly long https://jsonblob.com/1042496352801800192
```

### Device

Samsung Galaxy S8, Iphone 8

### VisionCamera Version

2.14.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

demsr, 2022-12-01:
If you collapse the "formats" array the output looks exactly like the one in the [docs](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDevice) (different order though). And yes, the formats array is quite long...

TideOPL, 2022-12-02:
Is the 'formats' array used for anything in particular?

demsr, 2022-12-02:
Yes, it lists all the available formats your cameras support. Read more about formats [here](https://mrousavy.com/react-native-vision-camera/docs/guides/formats)

TideOPL, 2022-12-02:
Ah right, so this issue is a bit of a non issue?

demsr, 2022-12-02:
From my point of view this is no issue, no.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy