margelo/react-native-vision-camera #1343: 🐛 React Native: Recording crash on few android mobiles due to  react-native-video used

ZakirBangash, 2022-11-17:
### What were you trying to do?

I want to record the video, although it is working on IOS and other mobiles but crash on infinix 12 and on Nokia 3.4 

React Native version: 0.64.0 React-native-vision-camera 2.13.0

### What happened instead?

There is no such log for crashing, but android studio shows that 

[image]

### Device

Infinix hot 12 and Nokia 3.4 

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ChetanShishodiaLiqvid, 2022-11-21:
We are also facing the same crash issue on recording on some android devices. Please suggest the needful steps to fix the issue.

ZakirBangash, 2022-11-22:
Could you tell us @ChetanShishodiaLiqvid which version of react native are you using? and also please let me know which version of react-native-vision camera are you using?

ChetanShishodiaLiqvid, 2022-11-22:
> 

@ZakirBangash 
"react-native": "^0.67.4",
 "react-native-vision-camera": "^2.13.3",
 
I also tried after upgrade all the packages with their latest versions, but no success.

ZakirBangash, 2022-11-23:
Same here, but somebody recommends that enable Hermes after that check the app, but I'm working on it, enabling Hermes giving me errors, I am on it and hope it works.

ChetanShishodiaLiqvid, 2022-11-23:
> @ChetanShishodiaLiqvid Same here, but somebody recommends that enable Hermes after that check the app, but I'm working on it, enabling Hermes giving me errors, I am on it and hope it works.

@ZakirBangash Thank you for the suggestion, I'll also try this.

ZakirBangash, 2022-11-25:
@ChetanShishodiaLiqvid let me know if it works for you or not

gokul-sirpi, 2023-07-23:
anyone has solution for this crashing issue?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy